greiman/SdFat #7: Issue opening files

Arianet, 2015-02-01:
Hi and many thanks for your work on this library.  I have been having some difficulties using the library and in my own code and the example code for read and write of a file am hitting the same problem in that opening files fails.  I have un-commented the fail macro & SD_TRACE serial print statements and the trace looks like this:

Press enter to start
Started
RB0
RB8192
RB12596
SD Init Complete
RB16384
..Arduino\libraries\SdFat\SdVolume.cpp224
..Arduino\libraries\SdFat\SdBaseFile.cpp1107
..Arduino\libraries\SdFat\SdBaseFile.cpp644
error: opening test.txt for write failed
SD errorCode: 0XF,0X1

Your assistance would be appreciated.
Many thanks, Bob.

Arianet, 2015-02-01:
SD Info Output:

SdFat version: 20141111

type any character to start

init time: 13 ms

Card type: SDHC

Manufacturer ID: 0X3
OEM ID: SD
Product: SL08G
Version: 8.0
Serial number: 0X90D2FE28
Manufacturing date: 11/2014

cardSize: 7948.21 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true
OCR: 0XC0FF8000
SD Partition Table
part,boot,type,start,length
1,0X0,0XB,8192,15515648
2,0X0,0X0,0,0
3,0X0,0X0,0,0
4,0X0,0X0,0,0
Volume is FAT32
blocksPerCluster: 64
clusterCount: 242304
freeClusters: 242300
freeSpace: 7939.69 MB (MB = 1,000,000 bytes)
fatStartBlock: 12596
fatCount: 2
blocksPerFat: 1894
rootDirStart: 2
dataStartBlock: 16384

greiman, 2015-02-01:
I can't help you since I don't have your hardware.  Your SD card sends an error token in response to reading block 16384.  This is the first block of the root directory.

closed 2015-02-01 by greiman