greiman/SdFat #8: Do the soft spi macros work for 1284?

bperrybap, 2015-02-17:
Was looking at the code today and saw what looks like a few issues in utility/DigitalPin.h
with respect to the 1284 support for for bobuino, mighty, and standard.
when using Jack's 1284 core package:
( https://github.com/JChristensen/mighty-1284p )

I'm assuming that this is a typo:
```#ifdef defined(VARIANT_MIGHTY)```

Then I had a question about the VARIANT_XXX defines being used to determine
which variant is being used. Where are those set? The IDE doesn't set them and they are not
being defined in the 1284 core variant/{variantname}/pins_arduino.h file.

Maybe I'm missing something but I don't see these being set anywhere.
My openGLCD library also does AVR raw port i/o so I also have to determine which
variant is being used. For the 1284 I use the values of analogInputToDigitalPin()
to adjust my own internal defines.
```
/*
 * Now must check which mighty1284p core variant
 * Look at the pin number of the first analog pin to distinguish
 */
#if (analogInputToDigitalPin(0) == 24 )
#define GLCD_CORE_MIGHTY1284P // avr mighty1284p "standard" pin mapping
#elif (analogInputToDigitalPin(0) == 31 )
#define GLCD_CORE_SANGUINO // avr mighty1284p avr_developers pin mapping (SANGUINO)
#elif (analogInputToDigitalPin(0) == 14 && analogInputToDigitalPin(7) == 21)
#define GLCD_CORE_BOBUINO // avr mighty1284p "bobuino" pin mapping
#elif (analogInputToDigitalPin(0) == 21 && analogInputToDigitalPin(7) == 31)
#define GLCD_CORE_SBEAUTY // "SleepBeauty" variant mighty-1284p core ------
#else
#warning "Uknown 644/1284 core: Using Arduino Core-Code Mode"
#define GLCD_CORE_CORECODE
#endif
```

greiman, 2015-02-17:
I will not be updating this version of SdFat.  You can define symbols like VARIANT_MIGHTY in pins_arduino.h and fix the typo in this version if you wish.

I will put your variant test in the new SdFat-beta and soon the beta will replace this version of SdFat.

closed 2015-02-17 by greiman

bperrybap, 2015-02-17:
"this version". I'm not understanding what that means.
I'm not seeing anything but the  master branch in this repo.
Do you have another repo somewhere?

So are you saying that in future releases the checks using VARIANT_XXX checks will
switch to using something that is available in the pins_arduino.h header file so it will
"automagically" work without users having to manually edit any files?

Today I'm going through all the 1284 code to verify that their pin definitions are consistent in all their
variants - they have had issues with this several times in the past and if there any issues the conditional checks above may change. I had to make a change to the checks yesterday since I had done that code in 2013 and they corrected an error in 2014 the a macro I use which affected the pin mappings and
I didn't see that change until a few days ago.

I only mention this as it is possible that the above conditionals may not be the final ones for
the 1284 core if I find any issues  in their mappings.

greiman, 2015-02-17:
This version means the SdFat repository.  I am about to totally replace the code in the SdFat repository with SdFat-beta here:

https://github.com/greiman/SdFat-beta
 
I have made the 1284 changes in the SdFat-beta repository so please test that code.  I no longer test on 1284 since there are so few users so let me know what you find.

bperrybap, 2015-02-17:
I'm curious, why a separate repo vs just creating a branch?
I can't fully test 1284 either. I only test Sleeping Beauty which is the only 1284 board I have.
I haven't played with SD cards yet on Arduino. I'm only jumped in here since some issues came up on one of the Arduino 1284 threads and I saw a few issues when I looked at the code.

greiman, 2015-02-17:
"I'm curious, why a separate repo vs just creating a branch?"

Most users just hit the "Download Zip" button so I have two repositories.

GitHub is not the real development site for the SdFat base code.  I don't use Arduino for my projects.  The base code is developed for use with RTOSs on ARM boards with SDIO and USB mass storage.

I don't plan to update SdFat very often in the future.

bperrybap, 2015-02-17:
If you want a git host that also supports download images, you could use BitBucket instead.
It allows multiple download images that users can select from and you also get download
statistics to see how many people have downloaded it.
Since you create the zip images, you can create the zip images
with a proper name and tree so users can use the Arduino IDE GUI to install the library -
something they can't do with a github repo zip image. 

I do this for my openGLCD library:
https://bitbucket.org/bperrybap/openglcd/wiki/Home

You can also use google code and then put the download images in the tree or in the wiki repo for
the project and then provide links to them on the main google code page or wiki page.
Google code used to have a downloads section, but they removed it, you get around this
by putting the images in the tree or on the wiki.
Google code can also host the documentation.

I will be switching back to google code for openGLCD as that way I can
host ready to install zip images, as well as the all the doxygen generated html library documentation.

I think it is very useful for users to be able to use the Arduino IDE GUI for installation
of libraries.
It also ensures that they don't screw up the installation.