greiman/SdFat #14: spi on other pin, Soft spi on Arduino ZERO

bobricius, 2015-06-29:
Hi I am try to compile your great library on my custom AtSamd21E18 zero compatible board
I have onboard SD card reader connected to MISO a19(d12),SCK a17(d13), MOSI a16(d11),SS a18(10)
I have try Soft spi because I do not know how to configure samd sercoms.
I get this erros:

In file included from D:\_sukromne\_Arduino\arduino-1.6.5\portable\..\..\Arduino\libraries\SdFat/utility/SoftSPI.h:31:0,
                 from D:\_sukromne\_Arduino\arduino-1.6.5\portable\..\..\Arduino\libraries\SdFat/SdSpi.h:211,
                 from D:\_sukromne\_Arduino\arduino-1.6.5\portable\..\..\Arduino\libraries\SdFat/SdSpiCard.h:29,
                 from D:\_sukromne\_Arduino\arduino-1.6.5\portable\..\..\Arduino\libraries\SdFat/SdFat.h:26,
                 from SoftwareSpi.ino:7:
D:\_sukromne\_Arduino\arduino-1.6.5\portable\..\..\Arduino\libraries\SdFat/utility/DigitalPin.h: In function 'bool fastDigitalRead(uint8_t)':
D:\_sukromne\_Arduino\arduino-1.6.5\portable\..\..\Arduino\libraries\SdFat/utility/DigitalPin.h:63:33: error: 'const PinDescription' has no member named 'pPort'
   return g_APinDescription[pin].pPort->PIO_PDSR & g_APinDescription[pin].ulPin;
                                 ^
D:\_sukromne\_Arduino\arduino-1.6.5\portable\..\..\Arduino\libraries\SdFat/utility/DigitalPin.h: In function 'void fastDigitalWrite(uint8_t, bool)':
D:\_sukromne\_Arduino\arduino-1.6.5\portable\..\..\Arduino\libraries\SdFat/utility/DigitalPin.h:73:28: error: 'const PinDescription' has no member named 'pPort'
     g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;
                            ^
D:\_sukromne\_Arduino\arduino-1.6.5\portable\..\..\Arduino\libraries\SdFat/utility/DigitalPin.h:75:28: error: 'const PinDescription' has no member named 'pPort'
     g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin;
                            ^
Error compiling.

greiman, 2015-06-29:
You will need to write a software SPI library for your board.

If you have ported the standard Arduio SPI library to your board use this define

#define SD_SPI_CONFIGURATION 1

From: Peter Misenko 
Sent: Sunday, June 28, 2015 11:51 PM
To: greiman/SdFat 
Subject: [SdFat] spi on other pin, Soft spi on Arduino ZERO (#14)

Hi I am try to compile your great library on my custom AtSamd21E18 zero compatible board
I have onboard SD card reader connected to MISO a19(d12),SCK a17(d13), MOSI a16(d11),SS a18(10)
I have try Soft spi because I do not know how to configure samd sercoms.
I get this erros:

In file included from D:_sukromne_Arduino\arduino-1.6.5\portable....\Arduino\libraries\SdFat/utility/SoftSPI.h:31:0,
from D:_sukromne_Arduino\arduino-1.6.5\portable....\Arduino\libraries\SdFat/SdSpi.h:211,
from D:_sukromne_Arduino\arduino-1.6.5\portable....\Arduino\libraries\SdFat/SdSpiCard.h:29,
from D:_sukromne_Arduino\arduino-1.6.5\portable....\Arduino\libraries\SdFat/SdFat.h:26,
from SoftwareSpi.ino:7:
D:_sukromne_Arduino\arduino-1.6.5\portable....\Arduino\libraries\SdFat/utility/DigitalPin.h: In function 'bool fastDigitalRead(uint8_t)':
D:_sukromne_Arduino\arduino-1.6.5\portable....\Arduino\libraries\SdFat/utility/DigitalPin.h:63:33: error: 'const PinDescription' has no member named 'pPort'
return g_APinDescription[pin].pPort->PIO_PDSR & g_APinDescription[pin].ulPin;
^
D:_sukromne_Arduino\arduino-1.6.5\portable....\Arduino\libraries\SdFat/utility/DigitalPin.h: In function 'void fastDigitalWrite(uint8_t, bool)':
D:_sukromne_Arduino\arduino-1.6.5\portable....\Arduino\libraries\SdFat/utility/DigitalPin.h:73:28: error: 'const PinDescription' has no member named 'pPort'
g_APinDescription[pin].pPort->PIO_SODR = g_APinDescription[pin].ulPin;
^
D:_sukromne_Arduino\arduino-1.6.5\portable....\Arduino\libraries\SdFat/utility/DigitalPin.h:75:28: error: 'const PinDescription' has no member named 'pPort'
g_APinDescription[pin].pPort->PIO_CODR = g_APinDescription[pin].ulPin;
^
Error compiling.

greiman, 2015-06-29:
You will need to write a software SPI library for your board. If you have ported the standard Arduio SPI library to your board use this define

closed 2015-06-29 by greiman