greiman/SdFat #16: support for arduino zero

ramonschepers, 2015-07-19:
seemingly there's no support for the arduino zero (arduino ide 1.6.5 r2), as i get alot of compile errors

SDFAT\SdSpiCard.cpp.o: In function `SdSpiCard::spiSend(unsigned char)':
C:\Users\Ramon\Documents\Arduino\libraries\SDFAT/SdSpiCard.h:263: undefined reference to `SdSpi::send(unsigned char)'
SDFAT\SdSpiCard.cpp.o: In function `SdSpiCard::spiInit(unsigned char)':
C:\Users\Ramon\Documents\Arduino\libraries\SDFAT/SdSpiCard.h:254: undefined reference to `SdSpi::init(unsigned char)'
SDFAT\SdSpiCard.cpp.o: In function `SdSpiCard::spiReceive()':
C:\Users\Ramon\Documents\Arduino\libraries\SDFAT/SdSpiCard.h:257: undefined reference to `SdSpi::receive()'
SDFAT\SdSpiCard.cpp.o: In function `SdSpiCard::spiReceive(unsigned char*, unsigned int)':
C:\Users\Ramon\Documents\Arduino\libraries\SDFAT/SdSpiCard.h:260: undefined reference to `SdSpi::receive(unsigned char*, unsigned int)'
SDFAT\SdSpiCard.cpp.o: In function `SdSpiCard::spiReceive()':
C:\Users\Ramon\Documents\Arduino\libraries\SDFAT/SdSpiCard.h:257: undefined reference to `SdSpi::receive()'
C:\Users\Ramon\Documents\Arduino\libraries\SDFAT/SdSpiCard.h:257: undefined reference to `SdSpi::receive()'
C:\Users\Ramon\Documents\Arduino\libraries\SDFAT/SdSpiCard.h:257: undefined reference to `SdSpi::receive()'
SDFAT\SdSpiCard.cpp.o: In function `SdSpiCard::spiSend(unsigned char)':

the list goes on, but i thought i'd save some reading.

greiman, 2015-07-19:
Try the beta.

https://github.com/greiman/SdFat-beta

closed 2015-07-19 by greiman

KingZogg, 2016-07-06:
Any word on this working for the Zero ?

greiman, 2016-07-06:
I don't have a Zero.  How does the beta fail on Zero?

KingZogg, 2016-07-06:
Hi,
Sorry.
I was using the old library. (Aduino newbie with libs etc)
It seems to be just fine.
The lowlatencylogger example is awesome !
Thank you so much for all your work.

rickyelqasem, 2018-05-18:
I cannot get the simple Longnames example working on a Zero. I know the hardware is ok because SD.h can read the SD card. I run the same code on a Mega and it works fine. Anything specific I have to do for a Zero?

greiman, 2018-05-18:
The LongFileName example works on my Zero.

Here is the compile/load:

```
Sketch uses 18216 bytes (6%) of program storage space. Maximum is 262144 bytes.
Open On-Chip Debugger 0.9.0-g932ec70 (2017-02-15-12:34)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000858 msp: 0x200023a0
** Programming Started **
auto erase enabled
wrote 24576 bytes from file C:\Users\bill\AppData\Local\Temp\arduino_build_543647/LongFileName.ino.bin in 2.671261s (8.985 KiB/s)
** Programming Finished **
** Verify Started **
verified 18488 bytes in 1.506024s (11.988 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```
I copied the test files to an 16 GB microSD and ran the example.  Here is the output:

```

You can use test files located in
SdFat/examples/LongFileName/testFiles
FreeStack: 29371

0 With.Two dots.txt
1 A long name can be 255 characters.txt
2 LFN,NAME.TXT
3 lower.txt
4 MIXCASE.txt
5 mixed.TXT
6 Not_8_3.txt
7 OK%83.TXT
8 STD_8_3.TXT
9 With Blank.txt

Enter File Number: 9

With Blank.txt
Just another example of a Long File Name.

Enter File Number:
```

rickyelqasem, 2018-05-18:
I'm using this board which is a Zero compatible  Nano board > https://wiki.protoneer.co.nz/NANO-ARM . Like I said I know SD card and files are ok because I SD.H can grab the files on the card. SdFat stops at begin()

greiman, 2018-05-18:
Can't help. There are hundreds of Arduino-like boards so I can't buy and test all.

rickyelqasem, 2018-05-18:
That's the problem... Mega == Work, Due == Work, Nano-ARM == not work... I have another SAMD21 en route from China so lets see that works.

rickyelqasem, 2018-05-18:
Add Nano to that list of working units... How frustrating. and weirdly SdInfo works using the default parameters. Where are you based Bill? I wonder if I can persuade the Nano-Arm guys to send you a board.

greiman, 2018-05-18:
SdInfo often works when other examples fail.  The reason is SdInfo does not initialize the SD until a character is typed.  This insures there is not a spurious restart related to the USB. 

A real Zero has an extra chip associated with the USB.

Try adding the following to the LongFileName example.

```
void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  delay(1000);

  Serial.println(F("Type any character to begin.")); // <<-- ADD THIS
  while (!Serial.available()) {}  // <<--ADD THIS
```

Also try the QuickStart example.

rickyelqasem, 2018-05-18:
Ok this is what tried:

- your added code above made no difference 
- when I tried the quickstart I got different results. If I placed in a FAT16 128mb SD card I got some feedback but no files displayed. When I put in a FAT32 16GB SD card it errored. See the output below.
-  these boards only have NativeUSB port so I have to change the Serial commands to SerialUSB < I wonder if your code is failing inside the library because it references Serial and not SerialUSB? clutching at straws but its an observation. 

**output from 128MB SD card**
Enter the chip select pin number: 4
Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.
Card successfully initialized.
Card size: 122 MB (MB = 1,000,000 bytes)
Volume is FAT16, Cluster size (bytes): 2048
Files found (date time size name):

**output from 16GB SD card**
SD initialization failed.
Do not reformat the card!
Is the card correctly inserted?
Is chipSelect set to the correct value?
Does another SPI device need to be disabled?
Is there a wiring/soldering problem?
errorCode: 0x43, errorData: 0xff
Restarting
Enter the chip select pin number:

rickyelqasem, 2018-05-18:
Bill. That was weird. I got it working.... I used SDformat to format the 128MB card. bear in mind that these cards worked no problem in Mega, Nano and Due... just not this NANO-ARM board... test the 16GB one next..

rickyelqasem, 2018-05-18:
Enough testing for 1 night... 
Mega/Due/Nano = working both 128mb FAT16 and 16GB FAT32 cards
Nano-ARM (Zero) = only working on 128mb FAT16 ... now that is the strangest thing right?

greiman, 2018-05-19:
>  I wonder if your code is failing inside the library because it references Serial and not SerialUSB? clutching at straws but its an observation.

Serial is accessed in sd.intitErrorHalt() and sd.errorHalt().  You can add an argument to use SerialUSB.  This should provide additional info about the failure.  Change the calls like this.

`    sd.initErrorHalt(&SerialUSB);`

`    sd.errorHalt(&SerialUSB, "open root failed");`

The big difference between the ten year old version of SdFat used in SD.h and the current SdFat is the SPI clock rate.  I used a very slow default rate in the first versions of SdFat since many SD modules had resistor level shifters for 5V to 3.3 V signal conversion and failed at higher speeds

SD.h still has a 4 MHz default rate.  The LongFileName example uses the max rate supported by the SPI library.  This is probably 12 MHz for the Nano Zero.

The SAMD implements SPI with the SERCOM interface so the signals may be marginal on the Nano Zero for some cards at 12 MHz.

Try 4 MHz in the examples.  Here is the change for LongFileName.

```
  // Limit SPI to 4 MHz clock
  if (!sd.begin(SD_CS_PIN, SD_SCK_MHZ(4))) {
    sd.initErrorHalt(&SerialUSB);
  }
```

rickyelqasem, 2018-05-19:
Bill its not your library that is the problem... is this board... the clue was in that it could read from small Sd cards but not larger ones and it gave me an idea. What if that was caused by not enough current getting to the SD module so I attached a separate power source and boom it worked...

greiman, 2018-05-19:
You got it.  I missed this on the board website.

> Power Management
> 
> 3.3V Regulation is done with a NCP551 Low-Dropout Voltage regulator.
> 
>     Input Voltage 12V max
>     Output currnet 150mA
>     Very low quiescent current of 4.0uA(Typical)

150 mA is way too little to power the SAMD and any other serious device.

The Zero has a 1 amp LDO 3.3 V regulator so you can get about 500 mA from USB and even more with an external PS since it has an efficient 1 amp DC-DC step-down regulator for the external jack. 

I also waste lots of time with poorly designed ebay and other Chinese boards and breakouts.  

I now do more testing after being burned too often.

rickyelqasem, 2018-05-19:
Yeah I picked that board because it don't look like a badly built clone. I have a low powered OLED to replace the LCD and if I can find a low powered SD module I can make it work. Thanks for your help BTW... Much appreciated. I was almost at the point where I would have purchased one of these and shipped it to your home :)

greiman, 2018-05-19:
>  I can find a low powered SD module I can make it work. 

That won't be easy.  Modern SD cards use high current for short times during flash access.  That's so they can be fast.  Average current is low but you need high peak current.