greiman/SdFat #17: LFN create/write

nostar, 2015-08-03:
Can you show an example of opening a new file for write (create) with long file name?

nostar, 2015-08-03:
Nevermind looks like I figured it out.  Using SdFile::open() instead of SdFat::open() lets my create my 18 character alphanumeric file.

greiman, 2015-08-04:
You should be able to create a long file name for write like this:

  myFile = SD.open("test long name.txt", FILE_WRITE);

Where SD is declared like this:

SdFat SD;

closed 2015-08-04 by greiman

nostar, 2015-08-04:
FYI that does not work for me.  No file is created unless the filename is 8.3 compliant.  What did work was SdFile::open() instead of SdFat::open() :

SdFat SD;
SdFile myfile;
SD.begin(SD_CS);
myfile.open("123456789012345678.bin", O_RDWR | O_CREAT) ;

greiman, 2015-08-04:
I downloaded the current zip for this repository.

I ran this program.   It printed "yes".  I found the file 123456789012345678.bin on the SD.
```
#include <SPI.h>
#include <SdFat.h>
#define SD_CS_PIN 10

SdFat SD;

File myFile;

void setup() {
  Serial.begin(9600);
  SD.begin(SD_CS_PIN);
  myFile = SD.open("123456789012345678.bin", O_RDWR | O_CREAT);
  myFile.close();
  if (SD.exists("123456789012345678.bin")) {
    Serial.println("yes");
  } else {
    Serial.println("no");
  }
}

void loop() {
}
```

nostar, 2015-08-05:
Oops.  I was using FILE_WRITE, not O_RDWR | O_CREAT, with SD.open().