greiman/SdFat #18: Missing SdVolume on SparkPlug MP3 2011 Code

promechitlungsei, 2015-08-04:
Hi I've searched on forums in Arduino and other web links online, but I'm am trying to test out the SparkPlug 2011 Code on my Arduino Nano, I've set the pins and included the correct libraries, but I'm missing SdVolume. I've looked at the text files of the changes and realized that it is deprecated, but then is there another function or even a library that I need to include for a function similar to SdVolume?

greiman, 2015-08-04:
I rewrote SdFat in 2011 and released a version in July 2011 with SdVolume  internal to The SdFat class.

I don't intend to support use of the SdVolume class or it's replacement, FatVolume in user programs.

There should be no need for these internal classes in  user programs.

closed 2015-08-04 by greiman

promechitlungsei, 2015-08-04:
Is there any way to download the 2011 version of the SdFat library?

greiman, 2015-08-04:
The 2011 versions no longer compile with current versions of the IDE.

The core of SD.h is a 2010 or 2011 version of SdFat.