greiman/SdFat #20: Importing the library correctly

steowens, 2015-09-28:
I am not exactly sure how to import this library into Arduino 1.6.5.  I tried copying the sdfat-master folder to the Libraries directory but that results in a compilation error unable to Locate Sdfat.h  What is the correct way to get this library working in Arduino 1.6.5 and why did the Arduino folks change things in a way that breaks legacy libraries?  (you don't need to answer that last, that is just me engaging in self pity).

millerlp, 2015-09-28:
Inside the SdFat-master folder that you got from github, you will find a
few other directories, including one labeled simply 'SdFat'. That is the
folder (the entire SdFat folder and its contents) that you need to put in
your Arduino/libraries/ directory.

AlexCalibur, 2015-11-09:
With Arduino 1.6.5, libraries can also be include in user space.
In your sketchbook, the is a special folder named 'libraries', copy SdFat dir at this place and restart Arduino IDE.

steowens, 2016-01-17:
Thanks for the info.

greiman, 2016-08-20:
Solved

closed 2016-08-20 by greiman