greiman/SdFat #21: can't remove an empty file in some case

lsaavedr, 2015-10-15:
Hi, surprising is not possible to remove an empty file, for example created as follows:

```sh
$ touch /pathToSD/Q0000000.QLS
$ umount /pathToSD
```

and then:

```c++
void setup() {
  while(!Serial.available());
  while(Serial.available()) Serial.read();

  SdFat sd = SdFat();
  sd.begin(SD_CS, SPI_FULL_SPEED);

  if (sd.remove("Q0000000.QLS")) {
    Serial.println("remove ok"); // is always true!!!
  } else {
    Serial.println("remove fail");
  }
}
void loop() {}
```
is amazing!, but if this method is unprivatized, make possible this:

```c++
void setup() {
  while(!Serial.available());
  while(Serial.available()) Serial.read();

  SdFat sd = SdFat();
  sd.begin(SD_CS, SPI_FULL_SPEED);

  if (sd.remove("Q0000000.QLS")) {
    Serial.println("remove ok"); // is always true!!!
  } else {
    Serial.println("remove fail");
  }

  {// some like flush SdFat
    sd.card()->chipSelectLow();
    sd.card()->waitNotBusy(SD_WRITE_TIMEOUT);
    sd.card()->chipSelectHigh();
  }
}
void loop() {}
```

this code remove the file, can some like flush() or end() be necessary to SdFat?

PD: if the file name is lower case or more length it work flawlessly...

greiman, 2015-10-15:
I have no idea why your program does not work.

I create the empty file Q0000000.QLS on a Windows 10 computer using an SD formatted with the SD association's SD formatter.

I run this program using the latest SdFat beta.
```
#include "SPI.h"
#include "SdFat.h"

void setup() {
  Serial.begin(9600);
  Serial.println("Type any character to remove file");
  while(!Serial.available());
  SdFat sd = SdFat();
  if (!sd.begin(10)) {
    Serial.println("begin Failed");
    return;
  }
  if (!sd.exists("Q0000000.QLS")) {
    Serial.println("file not found");
    return;
  }
  if (!sd.remove("Q0000000.QLS")) {
    Serial.println("Remove failed");
  } else {
    Serial.println("File Removed");
  }
}
void loop() {
}
```

The program prints
>Type any character to remove file
>File Removed

The file is not on the SD.

closed 2015-10-15 by greiman

lsaavedr, 2015-10-15:
I now understand the problem, but I don' know if you can reproduce this, the code is this:

```c++
#define SD_POW 14

#define SD_CS 10
#define SD_SCK 13
#define SD_MOSI 11
void setup() {
  // SD power off:
  pinMode(SD_CS, OUTPUT); digitalWrite(SD_CS, LOW);
  pinMode(SD_SCK, OUTPUT); digitalWrite(SD_SCK, LOW);
  pinMode(SD_MOSI, OUTPUT); digitalWrite(SD_MOSI, LOW);
  pinMode(SD_POW, OUTPUT); digitalWrite(SD_POW, LOW);

  while(!Serial.available());
  while(Serial.available()) Serial.read();

  digitalWrite(SD_POW, HIGH); // SD power on
  delay(200); // In SD-standard is 200ms, right? (this work fine)

  SdFat sd = SdFat();
  sd.begin(SD_CS, SPI_FULL_SPEED); // in this SD_CS, SD_SCK and SD_MOSI are reconfigured

  if (sd.remove("Q0000000.QLS")) {
    Serial.println("remove ok"); // is always true!!!
  } else {
    Serial.println("remove fail");
  }

  // SD power off:
  // delay(100); // What does the SD standard say about this? (with 100ms it work fine)
  pinMode(SD_CS, OUTPUT); digitalWrite(SD_CS, LOW);
  pinMode(SD_SCK, OUTPUT); digitalWrite(SD_SCK, LOW);
  pinMode(SD_MOSI, OUTPUT); digitalWrite(SD_MOSI, LOW);
  pinMode(SD_POW, OUTPUT); digitalWrite(SD_POW, LOW);
}
void loop() {}
```
the problem is caused by de-energizing the SD prematurely due to power control, with a delay(100) it work fine, but what does the SD-standard say about this?

greiman, 2015-10-15:
The standard says reliably removing power is not supported in SPI mode.  It does suggest that you can remove power one second after the card goes not busy but does not guarantee this will work.

You can check for busy like this.  This is used in several SdFat examples like the LowLatencyLogger at about line 441.
```
if (sd.card()->isBusy()) {
}
```
Also you should not power the card with a gpio pin.  Write can occasionally require more current than a pin can supply.  You may see data errors.

SdFat leaves cards in the mode where they can go into low current sleep mode.  For most cards sleep mode draws about 100 µA.

Users have powered a low power logger for up to a year with three AA batteries.

https://edwardmallon.wordpress.com/2014/07/01/a-10-diy-data-logger-is-born/

lsaavedr, 2015-10-15:
thanks!,

1) the SD_POW pin is connected to ~SHDN of the LT1962-3.3
2) with ```while (sd.card()->isBusy());``` instead of ```delay(100);``` it work!

lsaavedr, 2015-10-15:
the sandisk extreme pro:

https://www.sandisk.com/home/memory-cards/microsd-cards/extremepro-microsd

draws about 300µA in sleep mode

greiman, 2015-10-15:
You can't depend on isBusy() to power down a card.  It only means the card  can accept a command.  It may still be programming flash or moving data for wear-leveling.  You really need the one second delay after not busy.

The sandisk extreme pro may not be a good choice for SPI mode.  Often lower cost cards perform better in SPI mode.  SPI mode is an after thought for high end cards, required by the SD spec. Cards with good SPI performance often sleep at lower current.

Also, the trade-off between powering down a card and sleep is not simple.  Cards draw lots of current during initialization, often 100 times the sleep current.  Depends on the access pattern to the card and the time to initialize you card.