greiman/SdFat #22: Arduino mega 2560 problem with software spi

brcnpk, 2015-10-17:
Hi, I am trying to use MEGA with softwate SPI SdFatSoftSpi. However no luck. I have tried following example with UNO and MEGA (original hardware). All 3 modes (SdFatSoftSpi, SdFatLibSpi, SdFat) works fine with UNO. Only 2 modes  (SdFatLibSpi, SdFat) work with MEGA. I have tried 2 different shield on both. An ethernet shield with SD CS pin 4 and a sparkfun CANBUS shield with SD CS 9. Both the same result. I cannot use software SPI on MEGA. Please help. I am on IDE v1.6.5

Thanks,

/**************************************************************************/

#include <SPI.h>
#include <SdFat.h>
#if SD_SPI_CONFIGURATION >= 3  // Already set in SdFat/SdFatConfig.h

// Pin numbers in templates must be constants.
const uint8_t SOFT_MOSI_PIN = 11;
const uint8_t SOFT_MISO_PIN = 12;
const uint8_t SOFT_SCK_PIN  = 13;

// SD card Chip select 4 for ethernet, 9 for canbus shield
const uint8_t SD_CHIP_SELECT_PIN = 4;

// SdFat software SPI template (works with UNO only)
SdFatSoftSpi<SOFT_MISO_PIN, SOFT_MOSI_PIN, SOFT_SCK_PIN> sd;

// Using the standard Arduino SPI library (works with UNO and MEGA)
//SdFatLibSpi sd;

// Using  fast custom SPI (works with UNO and MEGA)
//SdFat sd;

// Test file.
SdFile file;

void setup() {
  pinMode(10,OUTPUT); // canbus and ethernet SPI CS.
  //digitalWrite(10,HIGH); 
  pinMode(53,OUTPUT); 
  //digitalWrite(53,HIGH);

  Serial.begin(9600);
  while (!Serial) {}  // Wait for Leonardo

  
  Serial.println("Type any character to start");
  while (Serial.read() <= 0) {}
  
  if (!sd.begin(SD_CHIP_SELECT_PIN)) {
    sd.initErrorHalt();
  }

  if (!file.open("SoftSPI.txt", O_CREAT | O_RDWR)) {
    sd.errorHalt(F("open failed"));
  }
  file.println(F("This line was printed using software SPI."));

  file.close();

  Serial.println(F("Done."));
}
//------------------------------------------------------------------------------
void loop() {}
#else  // SD_SPI_CONFIGURATION >= 3
#error SD_SPI_CONFIGURATION must be set to 3 in SdFat/SdFatConfig.h
#endif  //SD_SPI_CONFIGURATION >= 3

brcnpk, 2015-10-17:
I forgot to mention that the error is: it just cannot read the SD card.

Can't access SD card. Do not reformat.
No card, wrong chip select pin, or SPI problem?
SD errorCode: 0X1,0X0

greiman, 2015-10-17:
Your shields will not work with Software SPI and hardware SPI.  Both shields connect to the ISP connector so they use pins 50, 51, 52 for MISO, MOSI, SCK.

closed 2015-10-17 by greiman

brcnpk, 2015-10-17:
Hi Greiman,
First of all, canbus shield does not have ISP connector like the ethernet. It is connected via pins 9-13. Pin 9 is CS for SD and 10 is CS for CAN. Also, I stack mega+ethernet+canbus and use sofware SPI for CAN on pins 10-13 and hardware SPI for ethernet on pins 50-53 without any problems. In the mean time I can also switch to hardware SPI for SD card on ethernet shield. I want to use CAN shield SD card with mega. Because I will not use Ethernet shield, and want to use CAN + SD from pins 9-13 both with software SPI.

brcnpk, 2015-10-17:
Just to verify connectivity of SD pins: I used SdFatSoftSpi with ethernet shield on pins 50-53. No problem. Then, plugged in only CAN shield. Did the same, no luck. I am sure about CAN shield SPI pins that they are 9-13.

brcnpk, 2015-10-17:
> Bill Greiman <mailto:notifications@github.com>
> Saturday, October 17, 2015 3:14 PM
>
> Closed #22 <https://github.com/greiman/SdFat/issues/22>.

Hi Greiman,
Thanks for reply,

First of all, canbus shield does not have ISP connector like the 
ethernet. It is connected via pins 9-13. Pin 9 is CS for SD and 10 is CS 
for CAN. Also, I stack mega+ethernet+canbus and use sofware SPI for CAN 
on pins 10-13 and hardware SPI for ethernet on pins 50-53 without any 
problems. In the mean time I can also switch to hardware SPI for SD card 
on ethernet shield. I want to use CAN shield SD card with mega. Because 
I will not use Ethernet shield, and want to use CAN + SD from pins 9-13 
both with software SPI.

Just to verify connectivity of SD pins: I used SdFatSoftSpi with 
ethernet shield on pins 50-53. No problem. Then, plugged in only CAN 
shield. Did the same, no luck. I am sure about CAN shield SPI pins that 
they are 9-13.

greiman, 2015-10-17:
I have no idea whether you can run both the CAN and SD with software SPI on the CAN card.

You should be able to access the SD card alone if you disable the CAN by setting CAN_CS high. 

Looks like CAN_CS is 10 and in your example code above you set 10 to output mode but don't set it high.

Looks like SD_CS on your CAN shield is pin 9.

I test software SPI on mega using an Adafruit Dataloging shield with pins 10-13.  I use the SdFat SoftwareSpi example without modification for the test. The AdaFruit shield uses pin 10 for SD_CS.

Try making the following modes to the SoftwareSpi example.

Change the SD chip select pin like this
```
// Chip select may be constant or RAM variable.
const uint8_t SD_CHIP_SELECT_PIN = 9;
```
Disable CAN SPI like this
```
  pinMode(10, OUTPUT);  // <---  ADD THIS
  digitalWrite(10, LOW);  //  <<--- ADD THIS
  if (!sd.begin(SD_CHIP_SELECT_PIN)) {
    sd.initErrorHalt();
  }
```
Try running the example.

greiman, 2015-10-17:
I edited the above.  Make sure pin 10 is set high and only the CAN shield is on the Mega when you run the SoftwareSpi example.

brcnpk, 2015-10-19:
Here is the result;
2xWin7 PCs, 3xArduino IDEs with different versions with fresh install, Mega+CAN shield only, NO chance.
I switched CAN shield with another one. It worked!!!
However, I do not believe the first CAN shield is wrong because when I plug it on an UNO, with the same code above, it works in SdFatSoftSpi. 
Do you have any idea?

brcnpk, 2015-10-19:
More confusing finding:
I switched mega with a new one. Plugged in the problematic CAN shield. It worked..... ????

brcnpk, 2015-10-19:
By the way there is no hardware switch or jumper setting available on CAN shield and of course on MEGA. So hardware are all the same and originals from Sparkfun.

greiman, 2015-10-19:
I see cases like this fairly often.  Experienced users contact me when a problem doesn't make sense.

Often users finally understand the problem, like a bad solder joint on a board or different versions of a part on the board.

Sometimes you only know a board has a problem.  Like your Mega could have a problem with one of the software SPI pin connectors.  The new CAN shield might make connection because it's pins are at a slightly different angel.

Sometimes it's a mystery why two boards don't work together when they work in other situations.  If it is my hardware I can't rest until I understand this type problem.