greiman/SdFat #23: no known conversion for argument 1 from 'dir_t {aka directoryEntry}' to 'const dir_t* {aka const directoryEntry*}'

AlbanT, 2015-11-02:
When I try to compile the example ino from https://github.com/tilos/DueWebServer, which uses this SdFat library, I get the following error:

```
F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\SdFat/utility/FatFile.h:303:18: note:   no known conversion for argument 1 from 'dir_t {aka directoryEntry}' to 'const dir_t* {aka const directoryEntry*}'
Error compiling.
```
I'm having some trouble on pinpointing which library is giving me the problems :) 

For completeness I'll put the complete error below here:

```
F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\DueWebServer\DueWebServer.cpp: In function 'boolean WebServerHandler::get_handler(DueWebServer&)':
F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\DueWebServer\DueWebServer.cpp:533:47: error: invalid conversion from 'FatFile*' to 'SdBaseFile*' [-fpermissive]
     listDirectory(web_server, SdBaseFile::cwd());
                                               ^
In file included from F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\DueWebServer\DueWebServer.cpp:31:0:
F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\DueWebServer\DueWebServer.h:43:8: error:   initializing argument 2 of 'void WebServerHandler::listDirectory(DueWebServer&, SdBaseFile*)' [-fpermissive]
   void listDirectory(DueWebServer& web_server, SdBaseFile* file);
        ^
F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\DueWebServer\DueWebServer.cpp:544:34: error: cannot convert 'SdFile*' to 'SdBaseFile*' for argument '2' to 'void WebServerHandler::listDirectory(DueWebServer&, SdBaseFile*)'
   listDirectory(web_server, &file);
                                  ^
F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\DueWebServer\DueWebServer.cpp: In function 'void WebServerHandler::listFiles(const char*, SdBaseFile*, Client*, uint8_t)':
F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\DueWebServer\DueWebServer.cpp:632:26: error: no matching function for call to 'SdBaseFile::dirName(dir_t&, char [13])'
     file->dirName(p, name);
                          ^
F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\DueWebServer\DueWebServer.cpp:632:26: note: candidate is:
In file included from F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\SdFat/utility/ArduinoFiles.h:28:0,
                 from F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\SdFat/utility/FatLib.h:22,
                 from F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\SdFat/SdFat.h:27,
                 from F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\DueWebServer\DueWebServer.h:24,
                 from F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\DueWebServer\DueWebServer.cpp:31:
F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\SdFat/utility/FatFile.h:303:18: note: static uint8_t FatFile::dirName(const dir_t*, char*)
   static uint8_t dirName(const dir_t* dir, char* name);
                  ^
F:\Cloud\Box Sync\Arduino\Sketchbook\libraries\SdFat/utility/FatFile.h:303:18: note:   no known conversion for argument 1 from 'dir_t {aka directoryEntry}' to 'const dir_t* {aka const directoryEntry*}'
Error compiling.
```

greiman, 2015-11-02:
Looks like DueWebServer uses a very old version of SdFat.  Contact the author of the DueWebServer library.

closed 2015-11-02 by greiman

QGB, 2022-05-15:
```
remote Thread 1070150156 In: setup                                                        L58   PC: 0x42000072     at c:\Users\qgb\Documents\Arduino\libraries\SdFat\src/SdFat.h:83                                           #3  SdBase<FatVolume>::begin (maxSck=5000000, csPin=26 '\032', this=0x3fc8bc90 <sd>)                               at c:\Users\qgb\Documents\Arduino\libraries\SdFat\src/SdFat.h:74                                           #4  setup () at C:\Users\qgb\Documents\Arduino\Serial.println_delay/Serial.println_delay.ino:16                #5  0x42002834 in loopTask (pvParameters=<optimized out>)                                                          at C:\Users\qgb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\cores\esp32\main.cpp:42        #6  0x40386d2a in vPortSetInterruptMask ()                                                                         at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/r--Type <RET> for more, q to quit, c to continue without paging--c   
```

greiman, 2022-05-15:
I can't help, I no longer test SdFat on ESP32 since there are too many conflicts.