greiman/SdFat #25: Port for esp8266

Duality4Y, 2016-01-20:
what would it take to port this to esp8266 ?

https://github.com/esp8266/arduino

just wondering if it's not to work you could do it or I could do it.
but no ideas where to start to port it.

greiman, 2016-01-20:
Start with SdFat-beta.

Go to SdFat\src\FatLib\StdioStream.cpp:285 and make this change:
```
  //const char *p = (const char PROGMEM *)str;
  const char *p = (const char*)str;
```
Go to SdFat\src\FatLib\StdioStream.cpp:293 and make this change:
```
  //return p - (const char PROGMEM *)str;
  return p - (const char*)str;
```
Remove any calls to FreeStack() in the examples.  It is only an aid to tell how much free memory is on the stack.

It should compile.  I don't have hardware to test it so you will need to try it.

closed 2016-01-20 by greiman

Duality4Y, 2016-01-20:
that is so awesome will try it right away!

Duality4Y, 2016-01-20:
ok made the changes but it does not compile. i put the SdFat folder in SdFat-beta into the libraries folder under the sketchbook. and this is the compile output:
http://pastebin.com/KXraWQJf

greiman, 2016-01-20:
You did not install SdFat-beta correctly.
```
/home/robert/Sketchbook/sketchbook/libraries/SdFat/SdSpiCard.cpp:334: undefined reference to SdSpi::receive()
```
For SdFat-beta SdSpiCard.cpp is here:
```
SdFat/src/SdSpiCard/SdSpiCard.cpp
```
These errors will happen in SdFat-beta unless you remove FreeStack() from the bench example.
```
/tmp/Stino_build/bench/bench.ino.cpp.o:(.text.loop+0x14): undefined reference to `__brkval'
/tmp/Stino_build/bench/bench.ino.cpp.o:(.text.loop+0x18): undefined reference to `__bss_end'
```

Duality4Y, 2016-01-21:
Ok it compiles after i reinstalled the library.
but It keeps crashing.
```Use a freshly formatted SD for best performance.

Type any character to start

Soft WDT reset

ctx: cont 
sp: 3ffefca0 end: 3ffefeb0 offset: 01b0

>>>stack>>>
3ffefe50:  3fffdad0 00000000 3ffee908 4020239f  
3ffefe60:  3fffdad0 3ffee908 0000000a 40205384  
3ffefe70:  402010ae 00000000 3ffee908 402053dd  
3ffefe80:  3fffdad0 00000000 3ffee908 3ffeee7c  
3ffefe90:  3fffdad0 00000000 3ffeee74 402048dc  
3ffefea0:  feefeffe feefeffe 3ffeee90 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
here!

Use a freshly formatted SD for best performance.

Type any character to start

Soft WDT reset

ctx: cont 
sp: 3ffefc80 end: 3ffefeb0 offset: 01b0

>>>stack>>>
3ffefe30:  0000001c 3ffee908 0000000a 40205384  
3ffefe40:  00000020 3ffee908 40232b75 40204574  
3ffefe50:  3fffdad0 00000000 3ffee908 4020239f  
3ffefe60:  3fffdad0 3ffee908 0000000a 40205384  
3ffefe70:  402010ae 00000000 3ffee908 402053dd  
3ffefe80:  3fffdad0 00000000 3ffee908 3ffeee7c  
3ffefe90:  3fffdad0 00000000 3ffeee74 402048dc  
3ffefea0:  feefeffe feefeffe 3ffeee90 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
here!

Use a freshly formatted SD for best performance.

Type any character to start
```

Duality4Y, 2016-01-21:
ok got passed the crash by replacing a line of code where the while was waiting for input.

the esp8266 runs a operating system, and is not like a avr which does not.
when you take to much time away the operating system triggers a reset (watch dog timer reset)
giving back control to the operating system in tight loops you do a delay(0); or yield();
thus i replaced the waiting for input line with:
```
while (Serial.read() <= 0) {delay(0);}
```

after that it crashes with:
```
Use a freshly formatted SD for best performance.

Type any character to start

Soft WDT reset

ctx: cont 
sp: 3ffefc30 end: 3ffefeb0 offset: 01b0

>>>stack>>>
3ffefde0:  00002687 00000258 000000ff 4020293b  
3ffefdf0:  000000ff 0000002d 3ffeeba4 4020293b  
3ffefe00:  00001f5d 00000000 3ffeeba4 40202b0c  
3ffefe10:  00000000 00000000 3ffeeba4 40202953  
3ffefe20:  00001f5d 00001f5d 3ffeeba4 40202bb4  
3ffefe30:  00000002 3ffeee7c 40204894 3ffeee90  
3ffefe40:  402010ae 00000190 00000190 3ffeeba4  
3ffefe50:  3fffdad0 00000000 3ffee908 402023bf  
3ffefe60:  3fffdad0 3ffee908 0000000a 40205388  
3ffefe70:  402010ae 00000000 3ffee908 402053e1  
3ffefe80:  3fffdad0 00000000 3ffee908 3ffeee7c  
3ffefe90:  3fffdad0 00000000 3ffeee74 402048e0  
3ffefea0:  feefeffe feefeffe 3ffeee90 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

greiman, 2016-01-22:
Your on your own to fix the WDT problem.  The WDT timeout must be fairly long since begin() takes quite a while.

SdFat runs on sophisticated RTOSs like ChibiOS and FreeRTOS.  These are true preemptive systems.

SdFat has no delay() calls and no masking of interrupts since the OS should handle preemption.

Simple little systems like esp8266 are a pain.  I looked at the esp8266 code and wouldn't call it an OS. 

Try putting a yield() call in the long bench loops.

Here is what the esp8266 suggests.

```
If you have a loop somewhere in your sketch that takes a lot of time (>50ms) without
calling `delay`, you might consider adding a call to `delay` function to keep the WiFi
stack running smoothly.
```
There are no delay or yield calls int the ported SD.h library and that is just a very old version of SdFat.

You could try adding a yield call in SdSpiCard::waitNotBusy at about line 480 of SdSpiCard.cpp.  Just replace spiYield() with yield().

spiYield() is disabled and causes problems but esp8266 yield() might be OK.

Duality4Y, 2016-01-24:
ok i see :) I already put calls to yield in the forloops they run find, but they are slow! it slows it way down. best benchmark i got from it was 125KB/s

greiman, 2016-01-24:
The problem is the esp8266 system.  If I sprinkle yield calls in SdFat it will get slower.

The esp8266 SPI driver is slow.  You really can't use the fast mode since it requires data to be aligned on 32-bit boundaries.  

I use a Particle Photon for my WiFI development board.  They have a great SPI driver with DMA and good multi-threading.

https://www.particle.io/

Here are SdFat bench results for 512 bytes and 32KB on a Photon:
```
Buffer size 512 bytes

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
468.44,20400,577,1091

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
1315.36,1411,382,388

Buffer size 32768 bytes

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
2959.44,19153,10473,11054

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
3099.40,11962,10141,10575
```

Duality4Y, 2016-01-24:
those are some awesome results! :)
the spi on the esp8266 can go to 80MHz, there for I thought it would have been fast :s

greiman, 2016-01-24:
The esp8266 driver has huge overhead in the SPI and the system bus has the 32-bit alignment problem with it its FIFO.  The result is slow bye at a time programmed I/O. 

The STM32 on Photon does DMA which allows the system thread to run in parallel with the I/O.

Duality4Y, 2016-01-25:
It seems that I have allot of trouble compiling reliably.
one day it compiles and another it doesn't! maybe it's mysetup.

all I want though is to use long names. how hard would it be to hack that into the arduino SD library ?

greiman, 2016-01-26:
```
all I want though is to use long names. how hard would it be to hack that into the arduino SD library ?
does it rely on a lot of things?
``
I wrote the old version of SdFat in the Arduino SD library core in 2009.  I can't imagine hacking long file names into it.

Duality4Y, 2016-02-01:
I get the following error when I try to compile with your library:
/home/robert/Sketchbook/sketchbook/libraries/SdFat/examples/LongFileName/LongFileName.ino:93: undefined reference to `vtable for SdSpi'

any ideas what this possibly could be ?

greiman, 2016-02-01:
I can't help you.

I comment out the FreeStack stuff in the LongFileName example and compile it for a Adafruit HUZZAH ESP8266 Breakout.  I only get the expected warning about FreeStack().

> Build options changed, rebuilding all
> In file included from C:\Users\bill\AppData\Local\Temp\arduino_e32343f79511a425fb9487c3bdfa837b\LongFileName.ino:6:0:
> 
> C:\Users\bill\Documents\Arduino\libraries\SdFat\src/FreeStack.h:49:2: warning: #warning FreeStack is not defined for this system. [-Wcpp]
> 
>  #warning FreeStack is not defined for this system.
> 
>   ^
> 
> 
> Sketch uses 214,444 bytes (20%) of program storage space. Maximum is 1,044,464 bytes.
> Global variables use 34,006 bytes (41%) of dynamic memory, leaving 47,914 bytes for local variables. Maximum is 81,920 bytes.
>

Duality4Y, 2016-02-01:
that is verry weird I am compiling with esp generic board.