greiman/SdFat #28: Clearing the library's cache

Ivan-Perez, 2016-03-02:
Hello!

I've integrated the [SdFormatter example](https://github.com/greiman/SdFat/blob/master/SdFat/examples/SdFormatter/SdFormatter.ino) into my program. It does the following:

1. Print the root directory files, where there are 3-4 files.
2. Format the SD.
3. Print the root directory files again. At this moment no files should be shown, since the SD has been formatted, but instead the same list as in the first step is shown, I think because a cache issue.
4. Repeat step 3. At this moment the cache has been flushed internally and the listing prints no files (correct).

So my question is - is there a way to force clear the SD library internal cache?

Thanks for your great job!

greiman, 2016-03-02:
SdFormatter was not designed to do this.

You may want to look at this example.  It reformats an already formatted SD.

https://github.com/greiman/SdFat-beta/tree/master/SdFat/examples/wipe

From: Ivan-Perez 
Sent: Wednesday, March 2, 2016 2:27 AM
To: greiman/SdFat 
Subject: [SdFat] Clearing the library's cache (#28)

Hello!

I've integrated the SdFormatter example into my program. It does the following:

  1.. Print the root directory files, where there are 3-4 files. 
  2.. Format the SD. 
  3.. Print the root directory files again. At this moment no files should be shown, since the SD has been formatted, but instead the same list as in the first step is shown, I think because a cache issue. 
  4.. Repeat step 3. At this moment the cache has been flushed internally and the listing prints no files (correct). 
So my question is - is there a way to force clear the SD library internal cache?

Thanks for your great job!

greiman, 2016-03-02:
SdFormatter was not designed to do this. You may want to look at this example. It reformats an already formatted SD. 

[https://github.com/greiman/SdFat-beta/tree/master/SdFat/examples/wipe](url)

closed 2016-03-02 by greiman