greiman/SdFat #29: ADC drop outs with AnalogBinLogger

cmteuffel, 2016-03-04:
Hi,

first of all I have to admit that I did not grasp from the code how you got the adc values at all. I tried to use your library as a fast and cheap alternative for those very expensive digital data logger devices. What I've done is using your example out of the box with an UNO (and a Mega 2560) together with an Adafruit Logging Shield to log a (working) serial connection. Logging 5000 Samples per second worked well at first glance. However I could not completely reconstruct the data sent using the log file. Sometimes a high signal was simply missing. Curiously there were longer series of zero values recorded from time to time, while the program stated no error. Occasionally such an event happened when there was data to transmit. 

It seems to me that the data aquisition is broken. Reproduction should be easy as this happened with a standard setup. Enclosed some data samples that illustrate the frequency of such events. The red lines are adc readings that are exactly zero for more than one bit (1200 baud at 5000 samples).

Hopefully you can fix this, because it breaks the purpose of the logger.

Best regards,
cmteuffel

[image]
[image]
[image]
[image]

greiman, 2016-03-04:
I have no idea what you mean by logging a working serial connection.  What is your hardware setup?

I test by logging a sine wave and analyzing it.  See this DATA.png here:

[https://github.com/greiman/SdFat-beta/tree/master/AnalogBinLoggerExtras](url)

cmteuffel, 2016-03-04:
Thanks for your fast response. I'm aware that you've comprehensively tested the library and I read from your AdcErrorStudy.txt that you went quite some trouble to test the ADC. I simply used the library using the default settings in a real world hardware setup (e.g. USB powered UNO). 

Towards my hardware setup:
I have a serial loop back connection, simply receiving the sent data (ASCII) at the same PC. Therefore I know the connection is working and what's been sent when. I've put a resistor in the connection line and measure DC above this resistor while sending using the UNO contacts A0 and A1. 

Occasionally, the logger reads no voltage (series of zero values). At times, this drop out happens when there should be a voltage spike. So, while for example a character "1" (00110001) was sent and received using picocom, the same character is "!" (00100001) in the log file. This is, because one complete voltage spike was cancelled. These ADC drop outs seem to be randomly distributed. I could reproduce the behaviour with a Mega 2560. 

If this is because of my hardware setup, should'nt the drop outs also be visible at the terminal output? Maybe the default settings are the issue.

cmteuffel, 2016-03-04:
I just tried a different cabling and made another log attempt just to be sure. Curiously the drop outs occured again at precisely! the same positions.

errLocations1[1:10]
 52  79  88  95 149 251 359 372 421 427

errLocations2[1:10]
 52  79  88  95 149 251 359 372 421 427

greiman, 2016-03-04:
I setup a second Arduino with several programs that send serial.  I have no problems.

Here is a  typical program that sends a serial signal.

```
void setup() {
  Serial.begin(1200);
}
void loop() {
  Serial.write(0);
  Serial.write(1);
  Serial.write(2);
  Serial.flush();
  delay(10);
}
```
I connect digital pin 1 of the sending Arduino to analog pin 0 of the logging Arduino and log at 5,000 samples per second for a long time.

Here is a plot of a small part of one of the log files.

[image]
Note the 0, 1, 2 pattern.  This is Arduino 8-bit no parity serial.  I assume you understand serial data format.

I can't help you more since I can't reproduce your problem.

closed 2016-03-04 by greiman