greiman/SdFat #32: Help to create a new file in Sd2Card.

CoderSquirrel, 2016-05-12:
I used Marlin FIrmware to 3d printer, and i want to save a file on its sd and it use sdfat.
The card fortmat is Sd2Card.
I tried to follow and exemple that is there:
```
void CardReader::write_new_file(char *buf, char* filename) {
  SdFile myNewFile;
//   if (!myNewFile.open(root, filename, O_RDWR)) {
//    SERIAL_ECHO_START;
//    SERIAL_ECHOPGM(MSG_SD_CANT_ENTER_SUBDIR);
//    SERIAL_ECHOLN(filename);
//  }
  if (!myNewFile.open("teste.txt", O_CREAT | O_EXCL | O_WRITE)) {
    SERIAL_ECHO_START;
    SERIAL_ECHOPGM(MSG_SD_CANT_ENTER_SUBDIR);
    SERIAL_ECHOLN(filename);
  }
  char* begin = buf;
  char* npos = 0;
  char* end = buf + strlen(buf) - 1;

  myNewFile.writeError = false;
  if ((npos = strchr(buf, 'N')) != NULL) {
    begin = strchr(npos, ' ') + 1;
    end = strchr(npos, '*') - 1;
  }
  end[1] = '\r';
  end[2] = '\n';
  end[3] = '\0';
  myNewFile.write(begin);
  if (myNewFile.writeError) {
    SERIAL_ERROR_START;
    SERIAL_ERRORLNPGM(MSG_SD_ERR_WRITE_TO_FILE);
  }
}
```
And it didn't work.
So i want to know the correct way to write a file at Sd2Card because all the exemples i found was about sdfat card format.
Thanks

pavel-perina, 2016-06-16:
Same here, I've attached almost minimal sketch to reproduce bug. Might be in my code, perhaps I forgot to initialize something. file.open either resets arduino or it halts and initialization often takes 20-25ms (i'm used to 100ms) ... perhaps it depends if card was initalized already or disconnected from power before test). Anyways, file.open never returns.
I will try to copy sdout stream from getline example, at least example works.

Failing code: [sketch_jun17a.zip](https://github.com/greiman/SdFat/files/319463/sketch_jun17a.zip)
Warning: code sets pin 4 to output and high ... just in case you have something there

greiman, 2016-06-17:
pavel-perina,

Change this:
`  if (sd.cardBegin(SD_CHIP_SELECT, SPI_HALF_SPEED)) {`

To this:
`  if (sd.begin(SD_CHIP_SELECT, SPI_HALF_SPEED)) {`

And this will be the result:

> Boot up ...
INFO: sd.cardBegin() took 18ms
YYYY-MM.csv
File opened
Line printed
File written
OK

closed 2016-06-17 by greiman