greiman/SdFat #37: SoftSPI works with cheap card, not with expensive card

augustoborn, 2016-08-06:
Hi,

First of all, thanks for the lib!

I've been doing some raw block access using the Sd2Card class directly (and the writeBlock/readBlock methods). Everything works perfectly when I use the fast, custom SPI implementation (SD_SPI_CONFIGURATION = 0) and hardware SPI pins on the Arduino UNO.

Now I've moved my SD card to another SPI "bus" on pins A3 through A5, using SoftSpi (SD_SPI_CONFIGURATION = 2) to do so. Here's where it gets funky: 

1) Everything works fine on an old, cheap, no-name 128MB card;
2) On a nicer 8GB SanDisk Extreme Pro SDHC card, reading a block and querying the card for its size works;
3) On this faster card, writeBlock fails (error code SD_CARD_ERROR_WRITE).

Writing to this card works otherwise (hardware SPI on the Arduino or a PC), so I know it's fine. Could SoftSPI be too slow to write to this faster card?

Also, I'm at a loss as to how to debug this without tearing out my hair (I'm lacking an oscilloscope or logic analyzer), so any tips would be welcome.

Thanks!
Augusto

augustoborn, 2016-08-06:
Follow-up:

So I went and added longer delays beween clock edges in SoftSPI::sendBit and receiveBit (originally two nops), and writing to my faster SD card seems to work consistently after I add a (rather ridiculous) 100+ microsecond delay.

Chalk this one up to my setup? I'm on a breadboard, using a 74HC125 to level-shift, maybe the settling times on that are not quite industry standard.

greiman, 2016-08-07:
The best way to tune software SPI is with a scope.  software SPI can never equal the timing of the SPI controller.

Interrupts for other devices can cause errors.  At one point I disabled interrupts while a byte was transferred.  Most SD cards are very tolerant of SPI timing so I removed it.

I test software SPI with a variety of cards using the bench example and have not had a problem. 

I am working on a major rewrite of SdFat so I will look at software SPI timing again on a scope.

closed 2016-08-07 by greiman