greiman/SdFat #38: Hot plug/remove sample ?

luc-github, 2016-08-11:
Hi, very very nice library thanks for your great work. :smile: 
I am trying using it with my ESP8266 and have couple of question.

Is there an example of which functions need to be done when SD card is removed then plugged or even changed ?
Or just need to do a `SD.begin(SDCARD_CS_PIN);` again when card is inserted ?

Is there  a way that hot plug / remove,  can be safely detected using library functions, if there is no SDCARDDETECT pin ? Some card read don' have.
I tried  `SD.card()->cardSize()` and `File dir = SD.open("/"));` sometimes it works and sometimes not and even generate exception. 

Thanks in advance

greiman, 2016-08-11:
You must make sure all files opened for write are closed before removing an SD.

You can just call sd.begin() after the new SD is inserted.

There is no reliable way to detect insertion of an SD without a detect switch.  Many sockets have pull-ups or other problems so SdFat does not have a function to detect SD insertion.

closed 2016-08-11 by greiman

luc-github, 2016-08-11:
Thank you for your quick answer ^_^