greiman/SdFat #41: Problem with the writing time on the microSD using sdFat SoftwareSpi

JavernRey, 2016-08-20:
Hey!

First of all, thanks for your work!

I´m working on a project consisting on taking data from an accelerometer and saving it in a microSD card. I´m using an arduino UNO, an adafruit ADXL345 digital accelerometer, a Catalex microSD module and a Sandisk Ultra microSDHC card (8GB, class 10).

The accelerometer is using SPI, so I need to use the SoftwareSpi with the microSD module because I realised that Catalex module blocks the MISO pin, therefore it´s not possible to use it with other SPI devices at the same time. I followed the SofwareSpi example to do it.

I need a sampling frequency of 1000Hz and I managed to do it using the timerone library.
In order to check if it´s working properly, I save the microseconds each time the interruption is shot and I calculate the intervals between measures and the global frequency.
The accelerometer with timerone works fine and I always get intervals of 1000 microseconds. The problem is when I introduce the microSD module the intervals are not always 1000 microseconds. I have discovered that after a determinated amount of caracters it takes much more time to write the data.

I have tested it without using timerone library and without using the acelerometer, i.e. only writing on the microSD and calculating the intervals, and the result is the same:  after a determinated amount of caracters it takes much more time to write the data.

This is the code that I'm using. I have omitted the accelerometer section in order to simplify it, so now the accelerations for each axis x, y z are always equal to 10. In this example I measure 50 data (x, y, z).

`
#include <SPI.h>
#include <SdFat.h>
#include <TimerOne.h>

#if SD_SPI_CONFIGURATION >= 3  // Must be set in SdFat/SdFatConfig.h
#else  // SD_SPI_CONFIGURATION >= 3
#error SD_SPI_CONFIGURATION must be set to 3 in SdFat/SdFatConfig.h
#endif  //SD_SPI_CONFIGURATION >= 3

// Pin numbers in templates must be constants.
const uint8_t SOFT_MOSI_PIN = 5;
const uint8_t SOFT_MISO_PIN = 6;
const uint8_t SOFT_SCK_PIN  = 7;

// Chip select may be constant or RAM variable.
const uint8_t SD_CHIP_SELECT_PIN = 4;

// SdFat software SPI template
SdFatSoftSpi<SOFT_MISO_PIN, SOFT_MOSI_PIN, SOFT_SCK_PIN> sd;

// Test file.
SdFile file;

const unsigned long nData = 50;
float x = 10;
float y = 10;
float z = 10;
unsigned long t[nData];

volatile unsigned long t_i;
volatile bool measured;

void measure(void){
  t_i = micros();
  measured = true;
}

void save_microSD(float x, float y, float z){
    // Saving x
    String dataString_x = "";
    dataString_x = String(x);
    file.println(dataString_x);
    // Saving y
    String dataString_y = "";
    dataString_y = String(y);
    file.println(dataString_y);
    // Saving z
    String dataString_z = "";
    dataString_z = String(z);
    file.println(dataString_z);
}
  
void setup(){
  
  Timer1.initialize(1000);
  Timer1.attachInterrupt(measure);
  
  Serial.begin(250000);

  if (!sd.begin(SD_CHIP_SELECT_PIN)) {
    sd.initErrorHalt();
  }  
}

void loop(){

  if (!file.open("data.txt", FILE_WRITE)) {
    sd.errorHalt(F("open failed"));
  }
  
  Timer1.restart();
  
  for (int j = 0; j < nData; j++){
    measured = false;
    while (not measured){
    }
    save_microSD(x, y, z);
    t[j] = t_i;
  }
  
  Timer1.stop();
  file.rewind();

  //Print times to the terminal.
  for (int j = 0; j < nData; j++){
    Serial.print(j);Serial.print("T: "); Serial.print(t[j]); Serial.println("  ");
    if (j >= 1){
        Serial.print("Interval: "); Serial.print(t[j]-t[j-1]); Serial.println();
    }
  }
  unsigned long tiempo = t[nData-1]-t[0];
  Serial.print("Total time: "); Serial.println(tiempo);
  float f = ((nData-1)*1000000)/tiempo;
  Serial.print("Frequency: "); Serial.println(f);
  Serial.println();

  // Read microSD
  while (file.available()){
    Serial.write(file.read());
  }  
  file.close();

  if (sd.exists("data.txt")){
    sd.remove("data.txt") ;
  }
  
  delay(5000);  
}
`
The output that I obtain is the following:

0T: 5296956  
1T: 5297956  
Interval: 1000
2T: 5298956  
Interval: 1000
3T: 5299956  
Interval: 1000
4T: 5300956  
Interval: 1000
5T: 5301956  
Interval: 1000
6T: 5302956  
Interval: 1000
7T: 5303956  
Interval: 1000
8T: 5304956  
Interval: 1000
9T: 5305956  
Interval: 1000
10T: 5306956  
Interval: 1000
11T: 5307956  
Interval: 1000
12T: 5308956  
Interval: 1000
13T: 5309956  
Interval: 1000
14T: 5310956  
Interval: 1000
15T: 5311956  
Interval: 1000
16T: 5312956  
Interval: 1000
17T: 5313956  
Interval: 1000
18T: 5314956  
Interval: 1000
19T: 5315956  
Interval: 1000
20T: 5316956  
Interval: 1000
21T: 5317956  
Interval: 1000
22T: 5318956  
Interval: 1000
23T: 5319956  
Interval: 1000
24T: 5324956  
Interval: 5000
25T: 5325956  
Interval: 1000
26T: 5326956  
Interval: 1000
27T: 5327956  
Interval: 1000
28T: 5328956  
Interval: 1000
29T: 5329956  
Interval: 1000
30T: 5330956  
Interval: 1000
31T: 5331956  
Interval: 1000
32T: 5332956  
Interval: 1000
33T: 5333956  
Interval: 1000
34T: 5334956  
Interval: 1000
35T: 5335956  
Interval: 1000
36T: 5336956  
Interval: 1000
37T: 5337956  
Interval: 1000
38T: 5338956  
Interval: 1000
39T: 5339956  
Interval: 1000
40T: 5340956  
Interval: 1000
41T: 5341956  
Interval: 1000
42T: 5342956  
Interval: 1000
43T: 5343956  
Interval: 1000
44T: 5344956  
Interval: 1000
45T: 5345956  
Interval: 1000
46T: 5346956  
Interval: 1000
47T: 5347956  
Interval: 1000
48T: 5352956  
Interval: 5000
49T: 5353956  
Interval: 1000
Total time: 57000
Frequency: 859.00

Note that all intervals are 1000 microseconds except between data 23 - 24 and 47 - 48, which are 5000 microseconds. I don´t know why this is happening and if there is a way to get always the same writing time.

I hope you can help me with this problem. Thank you very much!

greiman, 2016-08-20:
You will never log to an SD at 1,000 Hz with this simple approach.

The long write times are when an actual SD block is written.

SD cards have 512 byte blocks and can take a very long time to write.  Often it will be much longer.  If a cluster must be added to the file, the FAT must be read and a cluster found. Two copies of the FAT must be updated.

The SD card can have occasional long write latency.  It must erase large blocks of flash.  It may pause to move data and remap flash for wear leveling.  This will take many ms.

Here is a program, [LowLatencyLoggerADXL345](https://github.com/greiman/SdFat-beta/tree/master/SdFat/examples/LowLatencyLoggerADXL345) that logs data from an ADXL345 at 500 Hz.  It require shared SPI and a high quality SD.

LowLatencyLoggerADXL345 appears to work at 1,000 Hz but may be marginal.

closed 2016-08-20 by greiman

JavernRey, 2016-08-22:
Hey!

Thank you very much for your help!

I will try this example. But I have one question. I also need to use a nRF24L01+ module to transmit data, which is working fine with ADXL345 using hardware SPI. Is it possible to use software SPI on the microSD module with this example? Or is it better to use hardware SPI on the microSD and software SPI on the ADXL345 and nRF24L01+?

Thank you very much!

greiman, 2016-08-22:
I can't help you.  The nRF24L01+ will fail if a TimerOne interrupt occurs while the nRF24L01+ is active.

I don't think you should try the LowLatencyLogger example if you want to add access to the nRF24L01+ while logging.

You are on your own.  I don't want to waste my time trying to solve this problem using Arduino.  

 I  would use an STM32 board with the ChibiOS/RT RTOS, not Arduino.   Even low cost STM32 boards have two or three hardware SPI ports.  The priority based preemptive scheduling of threads in ChibiOS/RT makse it easy to solve problems like this.

Good luck.

JavernRey, 2016-08-29:
Hey!

Just a short question, I don´t want you to waste your time.

I´m looking for a microSD module compatible with other SPI devices, i.e. which doesn´t hold the MISO line hostage. Could you tell me wich microSD module did you use for testing LowLatencyLoggerADXL345 or any other that I can use?

Thank you very much.

greiman, 2016-08-29:
I used an old Adafruit microsSD module.  I have some older ebay modules that are OK.

Now most ebay modules are clones of the flawed Catalex MicroSD Card Adapter or the LC studio module without level shifters

The Adafruit module is very expensive.

You may be interested in a test I did with an old TimerOne based program.

I use software SPI with an ADXL345.  See my ADXL345 driver in UserFunctions.cpp to define pins.

I have logged data directly to a csv file at 2,000 Hz with SdFat and at 4,000 with SdFatEX.

I have attached the program.  It is not polished since it is my test program.

[logger.zip](https://github.com/greiman/SdFat/files/442386/logger.zip)

JavernRey, 2016-08-29:
Hey!

Thank you for your answer. Now I´m not using TimerOne library. I have followed your program LowLatencyLoggerADXL345 and I have edited it, so now I can log data during the time that I set into a variable and then I can send it to a raspberry using nRF24L01+. It´s working fine, the problem is  that I´m using software SPI on the microSD and I can only get a frecuency of 200 Hz because if I try with higher frequency it says "Rate too fast". For this reason I´m going to buy the Adafruit microsSD module so I can use the hardware SPI on the three modules and I can get a frequency of 1000 Hz. I hope it will work.

Thank you very much!