greiman/SdFat #42: ESP8266 WIMOS pins

LaurentG91, 2016-09-22:
Dear developper,
I try to use the SdFat with a ESP8266 Wimos D1  and the SD Card at the same format.
The connexion pins used for this card is the following:

D5 =  GPIO14 = SCK         // SD reader 
D6 =  GPIO12 = MISO        // SD reader
D7 =  GPIO13 = MOSI        // SD reader
D8  = GPIO15, 10k Pull-down = SS // SD reader
I don't see any mean to define this pins to be used for the SdFat lib.

Any possible solution ?

Thanks 
Laurent

greiman, 2016-09-22:
SdFat uses the standard SPI library.  Just use the standard SPI pins.

closed 2016-09-22 by greiman

LaurentG91, 2016-09-23:
Sorry but this wiring is the standard wiring of ESP8266 WEMOS D1 based on ESP-12 and the standard WEMOS SD card produced by them. It's not a personal one...
http://www.wemos.cc/product/micro-sd-card-shield.html
I suppose your lib is not yet tuned for this card...
[image]

greiman, 2016-09-23:
Have you tried SdFat-beta.

I must warn you that I will likely drop SdFat support for ESP8266.

LaurentG91, 2016-09-23:
It's the 20160913 version.

greiman, 2016-09-23:
I checked several examples using SdFat-beta with an [Adafruit Huzzah ESP8226](https://www.adafruit.com/product/2471) and it seems to work.

LaurentG91, 2016-09-23:
Great.. I've check again and haven't found anything in your code.
So i've check the 2 cards and found à pb on a pin...
I just finish tested it on the train during travel and work fine on your
examples. Sorry for my questions... And Thanks a lot for this very good
development...
Thanks again