greiman/SdFat #46: License of SDFat?

jeremyherbert, 2016-10-12:
Hi,

I might be blind, and while I can see that SDFat is GPL as per the comments in the code, which version of GPL is it under? According to the Arduino folks, all of the core libraries are LGPL, and SD (based on this I believe) is part of the core library. Does that mean that SDFat is LGPL?

Thanks,
Jeremy

greiman, 2016-10-12:
SdFat is GPLv3.

The SdFat part of the SD library is GPLv3 also.  Read the comments .  It is clearly GPLv3.

From libraries/SD/src/utility/SdFat.h

/* Arduino SdFat Library
* Copyright (C) 2009 by William Greiman
*
* This file is part of the Arduino SdFat Library
*
* This Library is free software: you can redistribute it and/or modify
* it under the terms of the GNU General Public License as published by
* the Free Software Foundation, either version 3 of the License, or
* (at your option) any later version.

From: Jeremy Herbert 
Sent: Wednesday, October 12, 2016 4:39 AM
To: greiman/SdFat 
Subject: [greiman/SdFat] License of SDFat? (#46)

Hi,

I might be blind, and while I can see that SDFat is GPL as per the comments in the code, which version of GPL is it under? According to the Arduino folks, all of the core libraries are LGPL, and SD (based on this I believe) is part of the core library. Does that mean that SDFat is LGPL?

Thanks,
Jeremy

jeremyherbert, 2016-10-12:
Ok, thank you! I was confused because the arduino FAQ says that the core libraries are LGPL: https://www.arduino.cc/en/Main/FAQ

And SD is considered to be a core library here: https://www.arduino.cc/en/Reference/Libraries

closed 2016-10-12 by jeremyherbert

greiman, 2016-10-14:
SD is not a core library.  It is a "Standard Library".  Look at the various "Standard Libraries", few are LGPL.

Look at SD.h it is licensed GPLv3:

/*
 SD - a slightly more friendly wrapper for sdfatlib

 This library aims to expose a subset of SD card functionality
 in the form of a higher level "wrapper" object.

 License: GNU General Public License V3
          (Because sdfatlib is licensed with this.)