greiman/SdFat #47: SdFat Library changed dramatically?

drale2k, 2016-11-15:
Hi,

i am building an RFID reader (this one https://shubs.io/guide-to-building-the-tastic-rfid-thief/) but it seems that SdFat changed a lot since then. When i try to compile the project i get tons of error of missing files like "SDFile" and "SDVolume" which are no where to be found in the current repo.

I tried to find an older version but i am not able to. Should i use an old version or do i have to make changes in the project for the newer SdFat version to work?

Thank you (sorry for opening an issue for this but i don't where else to go)

hackgoblin, 2016-11-15:
I had the same issue with a different arduino build. These are the only copies of SDFat available. Google removed  the older files and this is the only repository.  I would update your code to accommodate the new files.

greiman, 2016-11-15:
I didn't save any old versions, so I can't help.  

The article refers to a five year old version of SdFat which probably will not work with current versions of the IDE.

closed 2016-11-15 by greiman

FernandoGarcia, 2016-12-12:
Hi @drale2k!

Try this version: https://github.com/FernandoGarcia/Ferduino_Portuguese/tree/master/Bibliotecas/SdFat

Best regards.