greiman/SdFat #50: Include SdFat library in other libraries not possible (?)

420579436, 2017-02-16:
I have started to replace the old SD library with the SdFat library in my libraries, but I get the error "Type 'SdFat' could not be resolved".

Example my_Library.h:
#include <SPI.h>
#include <SdFat.h>
SdFat SD;

Interestingly the SD declaration in .h file works in the main program but not in the .h or .cpp file.

Many thanks in advance.

greiman, 2017-02-16:
Probably the SdFat folder in not located in the libraries folder of the Arduino "Sketchbook" like this:
Arduino/libraries/SdFat

420579436, 2017-02-16:
I use the Eclipse 'Sloeber' IDE instead the Arduino IDE.
The SdFat library directory is in the same directory as e.g. the SPI library and Wire library directories.
If I include SdFat in the 'program' file, it works.
I have the problem only if I want to include the SdFat library in other libraries.

greiman, 2017-02-16:
I don't know how you cause third party libraries to be in the include list for 'Sloeber' IDE.

Eclipse has the ability to add an include path.

greiman, 2017-02-16:
http://eclipse.baeyens.it/how_to.shtml#/lib

420579436, 2017-02-16:
Eclipse has included the SdFat library and shows that the SdFat library has a problem to include the ArduinoFiles.h library.
The include SdFat.h works in a 'program' file, but not if included in another library.

420579436, 2017-02-16:
To add a library to Eclipse is very simple.
That is not the reason for the issue.

420579436, 2017-02-16:
I found another issue.
Maybe it helps.
if (!SD.exists("/MY_DIR")) { SD.mkdir("/MY_DIR"); } creates the file 'MY_DIR'!!
Only SD.mkdir("/MY_DIR"); creates the directory 'MY_DIR'.

greiman, 2017-02-16:
I don't use 'Sloeber' and can't help you.

420579436, 2017-02-16:
It's NOT a Eclipse/Sloeber problem!!!

Question:
Why works 'SdFile my_file;' and 'SdFat SD;' not in another library?

greiman, 2017-02-16:
This program seems to work fine.

```
#include <SPI.h>
#include "SdFat.h"
SdFat SD;
void setup() {
  Serial.begin(9600);
  Serial.println("Type any character");
  while (!Serial.available());
  if (!SD.begin()) {
    Serial.println("begin");
    while(1);
  }
  Serial.println(SD.exists("/MY_DIR"));
  if (!SD.exists("/MY_DIR")) { SD.mkdir("/MY_DIR"); }
  Serial.println(SD.exists("/MY_DIR"));
  Serial.println("Done");
}
void loop() {}
```

> 
> Type any character
> 0
> 1
> Done

420579436, 2017-02-16:
As I already mentioned.
The SdFat library works in a program but not in another library

420579436, 2017-02-16:
P.S.
Eclipse/Sloeber and Arduino use the same compiler.

greiman, 2017-02-16:
Ask a Sloeber user about this.  The include path can be a problem when you include a library in a library.

For example, if you want to include ArduinoFiles.h

#include "ArduinoFiles.h"

May not work since the file is located at" 

libraries/SdFat/src/FatLib/ArduinoFiles.h

greiman, 2017-02-16:
#include "SdFat.h"
#include "ArduinoFile.h"
void setup() {
}
void loop() {
}

Arduino: 1.8.1 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\bill\Documents\ArduinoSdFs\sketch_feb16a\sketch_feb16a.ino:2:25: fatal error: ArduinoFile.h: No such file or directory

 #include "ArduinoFile.h"

greiman, 2017-02-16:
This works with the Arduino IDE.

#include "SdFat.h"
#include "FatLib/ArduinoFiles.h"
void setup() {
}
void loop() {
}

stevstrong, 2017-02-16:
Instead
'SdFat SD;'
use
`SdFat my_sd;`

420579436, 2017-02-16:
> _Ask a Sloeber user about this. The include path can be a problem when you include a library in a library.
For example, if you want to include ArduinoFiles.h
#include "ArduinoFiles.h"
May not work since the file is located at"
libraries/SdFat/src/FatLib/ArduinoFiles.h_

Dear William,
Many thanks. I have written a couple of libraries for sensors and in this libraries I have included and included other libraries without problems.
The situation is, I want only to replace the old SD library with your much better SdFat library.
My libraries work all with the old SD library but the one by one replacement with your library is very difficult for me.
Thorsten

420579436, 2017-02-16:
#include "SdFat.h"
#include "ArduinoFiles.h"
void setup() {
}
void loop() {
}

Arduino: 1.8.1 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\bill\Documents\ArduinoSdFs\sketch_feb16a\sketch_feb16a.ino:2:25: fatal error: ArduinoFiles.h: No such file or directory

#include "ArduinoFiles.h"

420579436, 2017-02-16:
> _Arduino: 1.8.1 (Windows 10), Board: "**Arduino/Genuino Uno**"_

Dear William,
Many thanks for the important information.
You use a Uno and I use a Due.
We are using different compiler!

420579436, 2017-02-16:
> _I found another issue.
Maybe it helps.
if (!SD.exists("/MY_DIR")) { SD.mkdir("/MY_DIR"); } creates the file 'MY_DIR'!!
Only SD.mkdir("/MY_DIR"); creates the directory 'MY_DIR'._

This issue I don't have with an AVR board like an Uno.
This issue I have with a SAM board like a Due.
Thorsten

420579436, 2017-02-16:
> _Instead
SdFat SD;
use
SdFat my_sd;
because SD may be reserved._

Many thanks stevstrong.
A very good tip.
But unfortunately the beast (issue) is still alive.
Thorsten

greiman, 2017-02-17:
You expect me to guess you are using a Due?  

I don't appreciate your all caps AND !!! 

> It's NOT a Eclipse/Sloeber problem!!!

That does it I don't want to waste more time.

Please learn how to use Eclipse, CDT, and Sloeber.  Eclipse and CDT have excellent documentation. 

You need understand how gcc finds include files.  It's not AVR vs. SAM.  It's not the compiler.

Some libraries will work when included in a simple way.  Some won't.  SdFat has a more complex directory structure.   The Sloeber plugin may not discover include files in a sub-directory of the 'src' folder.

I think there is a YouTube video about the include browser.

You should only need to include SdFat.h.  SdFat.h includes other needed .h files.

You can always tweak the include paths in CDT.  Please read the documentation.

Avr and Due both fail in this example as does STM32, Teensy 3.x, and ESP8266.

```
#include <SPI.h>
#include "SdFat.h"
#include "ArduinoFiles.h"
void setup() {}
void loop() {}
```

> Arduino: 1.8.1 (Windows 10), Board: "Arduino Due (Programming Port)"
> 
> C:\Users\bill\AppData\Local\Temp\arduino_modified_sketch_210703\HelloWorld.ino:3:26: fatal error: ArduinoFiles.h: No such file or directory
> 
>  #include "ArduinoFiles.h"

This works on avr, Due, and other boards!  
```
#include <SPI.h>
#include "SdFat.h"
SdFat SD;
void setup() {
  Serial.begin(9600);
  Serial.println("Type any character");
  while (!Serial.available());
  if (!SD.begin()) {
    Serial.println("begin");
    while(1);
  }
  Serial.println(SD.exists("/MY_DIR"));
  if (!SD.exists("/MY_DIR")) { SD.mkdir("/MY_DIR"); }
  Serial.println(SD.exists("/MY_DIR"));
  Serial.println("Done");
}
void loop() {}
```

> Type any character
> 0
> 1
> Done

420579436, 2017-02-17:
No comment
Issue closed unsolved

closed 2017-02-17 by 420579436