greiman/SdFat #52: read file after restart esp8266

akaJes, 2017-03-09:
this issue appearing only when esp8266 device restarted without card power off and disappear when card reinserted before device restarting
i suppose it can depend of USE_MULTI_BLOCK_IO option enabled
project compiled with https://github.com/esp8266/Arduino 2.3.0 (with embeded library it works properly)
and used https://github.com/esp8266/Arduino/blob/2.3.0/libraries/ESP8266WiFi/src/WiFiClient.h#L76
write function which request from sd card blocks with size:
`#define WIFICLIENT_MAX_PACKET_SIZE 1460`
i will try to disable this option USE_MULTI_BLOCK_IO manually and test project again

attached file has first 512 bytes correct - then trash (it changes every time when i reload page) - then everything repeated
[image]

akaJes, 2017-03-09:
my code based on https://github.com/esp8266/Arduino/blob/2.3.0/libraries/ESP8266WebServer/examples/SDWebServer/SDWebServer.ino
just simple web server and used your library instead embeded based on your old/stable version https://github.com/esp8266/Arduino/tree/2.3.0/libraries/SD

this line https://github.com/esp8266/Arduino/blob/2.3.0/libraries/ESP8266WebServer/examples/SDWebServer/SDWebServer.ino#L85 generate result posted above
call stack:
https://github.com/esp8266/Arduino/blob/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h#L124
https://github.com/esp8266/Arduino/blob/2.3.0/libraries/ESP8266WiFi/src/WiFiClient.h#L76

greiman, 2017-03-09:
I don't debug user code.  Provide a simple example with the minimum code that demonstrates the problem.

If possible use [SdFat-beta](https://github.com/greiman/SdFat-beta) since it will soon replace this version of SdFat.

akaJes, 2017-03-13:
after manual disabling  `USE_MULTI_BLOCK_IO` in config, this issue disappeared !
i will try to use https://github.com/greiman/SdFat-beta

closed 2017-03-13 by akaJes

akaJes, 2017-03-22:
beta has the same issue , after disabling `USE_MULTI_BLOCK_IO` works properly

greiman, 2017-03-23:
I can't reproduce your problem with my tests on a Adafruit HUZZAH ESP8266.  I try various transfer sizes and verify the data.  I test at 80 and 160 MHz.

Once again, you must submit an example that isolates the problem in a reproducible way.  

I won't debug the web-server code since it doesn't error check each file operation for errors. 

Multi-block I/O can fail if there is any noise on the SPI bus due to wiring problems.   This could happen with ESP8266 since  yield() is called between SD blocks and noise from wi-fi could cause problems.