greiman/SdFat #54: Wierd Issue observed when reading files with filenames of 13 with same name as a subfolder in the volume working directory.

paqwa, 2017-04-15:
Hi Bill, 

I've been using your excellent sdFat Library (beta version for sdioEx downloaded from your github last week) to access the builtin memory card on a Teensy 3.6 and it works very well indeed. I'm working with this for integration with a TFT display so I've to read bmp's from time to time and I use the function attached to read the bmp's. I've made a slight alteration to the function from original source just to omit colors below certain threshold, i.e. similar to allowing for a transparent background which I find useful.

[sampe_code.txt](https://github.com/greiman/SdFat/files/923124/sampe_code.txt)

Now to the issue I'm experiencing with the sdFat library. The attached bmpDraw function appears to work excellently with your library. It is capable of opening and reading all information required from 24bit bmp's with either short or long filenames and I've verified both. However, I've come across a strange exception whereby the file isn't reading correctly and I really hope you can help me out which. It occurs under fairly specific conditions as follows.

1) Only occurs when working with a file with a long filename, i.e. abcdefghijkl.bmp
 works fine but abcdefghijklm.bmp doesn't read correctly.
2) Only occurs when a sub-folder of the same name as the the bitmap I'm trying to access exist together within the same working directory.  

The failure is occurring when trying to read the first few bytes in the file just after opening it with the statement:
 if (read16(bmpFile) == 0x4D42) // BMP signature
This condition isn't satisfied and so the bmpDraw function is bypassed. Again I note is was perfectly satisfied with the exact same file when no folder of the same name exists in the working directory, or if the filename length is reduced by one character and a folder with the same name exists.

This is all a bit weird for me and I'm banging my head off a brick wall at this stage but can't seem to find any possible causes, any idea's, possible causes, suggestions would be much appreciated. 

Regards,
Pat

greiman, 2017-04-15:
I am not sure what this means:

> Only occurs when a sub-folder of the same name as the the bitmap I'm trying to access exist together within the same working directory.

You can't have a sub-folder and file with the same name in a folder.

paqwa, 2017-04-16:
Hi Bill, thanks for your response. Sorry for not being clearer when describing the issue. Of course you cannot have two files/folders with exactly the same name in the same folder. What I meant was the file and folder have the same name when you exclude the file extension. With the extension included the filename is different to the folder. As an example the folder could be called:

abcdefghijklmno

and the file called

abcdefghijklmno.bmp

I hope this makes sense?

greiman, 2017-04-17:
Try this test program.  When I run it it prints this:

```
Type any character
         0 abcdefghijklmno/
       156 abcdefghijklmno.bmp
Done
```
```
#include <SPI.h>
#include "SdFat.h"

#define FOLDER_NAME "abcdefghijklmno"
#define FILE_NAME "abcdefghijklmno.bmp"

//SdFatSdio SD;
SdFatSdioEX SD;

File file;

void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  Serial.println("Type any character");
  while (!Serial.available());
  if (!SD.begin()) {
    Serial.println("begin failed");
    return;
  }
  if (!SD.exists(FOLDER_NAME)) {
    if (!SD.mkdir(FOLDER_NAME)) {
      Serial.println("mkdir failed");
      return;
    }
  }
  if (!SD.exists(FILE_NAME)) {
    if (!file.open(FILE_NAME, O_WRITE | O_CREAT | O_TRUNC)) {
      Serial.println("open write failed");
      return;
    }
    for (int i = 100; i < 256; i++) {
      file.write(i);
    }
    file.close();
  }
  if (!file.open(FILE_NAME, O_READ)) {
    Serial.println("open read failed");
    return;
  }
  for (int i = 100; i < 256; i++) {
    if (file.read() != i) {
      Serial.println("compare failed");
      return;
    }
  }
  SD.ls(LS_SIZE);
  Serial.println("Done");
}
void loop() {}
```

paqwa, 2017-05-04:
Hi Bill, yes this works for me also and thanks for putting it together. It prompted me to re-analyse my code in greater depth. I wrote my code to do different things with the images depending on whether a folder of the same name existed or not and in that section of the code I made a two very rookie mistakes! I didn't allocate a long enough char array for those filenames and upon analyzing my images on the SD card I found that additional information was saved to the file header which was not being skipped over by my code, hence the read error on images.

paqwa, 2017-05-04:
So it turns out my issue was just with my novice coding abilities and not with your excellent library, thanks all the same for your responses and guidance on the matter. I will close this issue now. 

All the best, 
Pat

closed 2017-05-04 by paqwa