greiman/SdFat #55: SD fat on Software SPI not compiling on MightyCore 1284p

SrikanthBala, 2017-04-22:
At the outset thank you for providing SDFat. I am currently using a MightyCore for 1284P and have successfully tested a SD card using your library on the 1284P running at 3.3V on a external 8MHz crystal on a breadboard using hardware SPI.
However when I used the code provided by you for SoftwareSPI (provided actually for 2560) my code first does not compile at all. and gives an error as below :

In file included from F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdSpiCard/DigitalPin.h:45:0,
                 from F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdSpiCard/SoftSPI.h:31,

                 from F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdSpiCard/SdSpi.h:235,

                 from F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdSpiCard/SdSpiCard.h:29,

                 from F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdFat.h:27,

                 from F:\arduino-1.8.2\portable\sketchbook\Series2_Tx\DumpInfo\DumpInfo\SDSPI1284\SDSPI1284.ino:7:

F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdSpiCard/boards/GpioPinMap.h:40:2: error: #error Undefined variant 1284, 644, 324
#error Undefined variant 1284, 644, 324

I have not got this error when using other libraries on 1284P. 

Finally I got the code to compile without errors by commenting out everything in GpioPinMap.h and adding the line #include "BobuinoGpioPinMap.h" as I am using the Bobuino pinout. After uploading, I realized my card is not recognized. 

I need to use SoftwareSPI as my RC522  with the SD card if both are attached to the same SPI bus. 

Kindly provide me with help as to why it does not compile in the first place without modifying Gpio header file and also why it doesnt recognize my card

I am using a SD fat library which says #define SD_FAT_VERSION 20160719
The Mighty Core is also recent I believe as I am using the IDE version 1.8.2 and downloaded it using Boards Manager. I am using Mighty Core MCUdude version 1.0.8

Please help

greiman, 2017-04-22:
I don't have a 8 MHz MightyCore  so I can't help you.

SrikanthBala, 2017-04-22:
Thank you sir for your prompt response. I am designing a new device which requires more pins than a ATMega 328. Hence I chose the 1284P. The other option I have is a Maple Mini like STM32 board. It is fast, has the pins I need and is cost effective. 
However I tried SdFat today on SPI1 using a LC MicroSd Shield bypassing the 5V regulator and using direct 3.3 V. However the level converter IC is not bypassed (its datasheet mentions it will work correctly for all voltages above 2.6v)

However the STM32 card returns failure to find the card using the SdInfo.ino. Pin connections made for SP1 and checked multiple times. No sketch works for SD card indicating a  basic fault in my system. Does your SdFat support STM32 Core ?

greiman, 2017-04-22:
Arduino STM32 requires the old Arduino IDE 1.6.12.  

SdFat no longer works reliably on STM32 and I don't want to fix it for 1.6.12.

SrikanthBala, 2017-04-22:
Ok

stevstrong, 2017-04-23:
A short comment:
Arduino_STM32 code works with both SdFat and SdFat beta, with Arduino 1.6.9 or later.
**But:** it only works with SD card modules designed for 3.3V, without 5V->3.3V regulator, see:
http://www.stm32duino.com/viewtopic.php?f=13&t=20, section "Known problems".

SrikanthBala, 2017-04-24:
Thank you stevstrong. 
I closed this issue earlier thinking I should just give up on the 1284p and the Mighty Core for Software SPI. I just got back here to see your comment. I AM using a 3.3V modified microSD module (as my 1284p is also running at 3.3V). I'll definitely give it a go with my Black Pill STM32 board. 

Meanwhile I reopened this because I finally found the problem with my code for the 1284P. 
After nearly 24 hrs of breaking my head on multiple SD card adapters and finally fixing one 'impossible to see' internally broken jumper wire, I found in a 2012 Arduino forum that SdFat SoftwareSpi assumes a Standard or Sanguino pinout and not Bobuino pinout. So I switched to Standard pinout names in the code and BINGO ! it works !
In fact it does not matter whether I select Bobuino board or Standard board under the boards selection for the Mighty Core. It still works on the Standard Pin names only as if the Standard Pins were somehow hardcoded into the SDFat SoftwareSPI code ignoring the definitions in Gpio.h. I dont know how to modify SDFat to do use the right pins so I'll leave it at that as I know how to get the the code to work.

Thanks Greiman for the library once again. ANd incidentally I tried it at both 16MHz and 8Mhz too and it works GREAT !
 I hope this helps someone else struggling on similar lines. Bye

closed 2017-04-27 by SrikanthBala