greiman/SdFat #56: Upgrade from 2015-03-24 version to 1.0.1 - enabling CRC on Mega2560 compile fails

AK6DN, 2017-04-29:
As the title says, I was previously using SdFat commit from 2015-03-24 with the following config:

#define USE_SD_CRC 2

but now when I update to v1.0.1 and edit SdFafConfig.h to set the CRC flag to 2 (or 1) the compile fails.

Error window:

`Arduino: 1.8.2 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Don\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10802 -build-path C:\Users\Don\AppData\Local\Temp\arduino_build_850350 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose E:\Projects\DEC_PDP_stuff\rx02_emulator\rx02_emulator\rx02_emulator.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Don\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10802 -build-path C:\Users\Don\AppData\Local\Temp\arduino_build_850350 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose E:\Projects\DEC_PDP_stuff\rx02_emulator\rx02_emulator\rx02_emulator.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
Generating function prototypes...
Compiling sketch...
Compiling libraries...
Compiling library "SPI"
Compiling library "SdFat"

C:\Users\Don\Documents\Arduino\libraries\SdFat\src\SdCard\SdSpiCard.cpp: In member function 'bool SdSpiCard::readData(uint8_t*, size_t)':

C:\Users\Don\Documents\Arduino\libraries\SdFat\src\SdCard\SdSpiCard.cpp:386:11: error: 'SD_CARD_ERROR_READ_CRC' was not declared in this scope

     error(SD_CARD_ERROR_READ_CRC);

           ^

Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Using library SdFat at version 1.0.1 in folder: C:\Users\Don\Documents\Arduino\libraries\SdFat 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.`

AK6DN, 2017-04-30:
Looks like in the new version of SdInfo,h that the def for SD_CARD_ERROR_READ_CRC was left out ...

```
  // Read/write errors
  SD_CARD_ERROR_READ = 0X50,
  SD_CARD_ERROR_READ_FIFO,
  SD_CARD_ERROR_READ_REG,
  SD_CARD_ERROR_READ_START,
  SD_CARD_ERROR_READ_TIMEOUT,
  SD_CARD_ERROR_STOP_TRAN,
  SD_CARD_ERROR_WRITE,
  SD_CARD_ERROR_WRITE_FIFO,
  SD_CARD_ERROR_WRITE_START,
  SD_CARD_ERROR_WRITE_TIMEOUT,

```

greiman, 2017-07-04:
Added SD_CARD_ERROR_READ_CRC to version 1.0.3.

closed 2017-07-04 by greiman