greiman/SdFat #57: cardSize Failed

schlaegerz, 2017-05-31:
I'm running the sdcard info example and getting a cardSize failed error
I'm not sure what this means.

SdFat version: 1.0.1

Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.

Assuming the SD chip select pin is: 15
Edit SD_CHIP_SELECT to change the SD chip select pin.

type any character to start
error: cardSize failed
SD errorCode: 0X52,0X7F

pmdmartins123, 2017-06-01:
On the sdfat library folder there is .h file that have the error codes description. When i locate the file i will post here again. If someone finds it out please post here.

greiman, 2017-06-01:
SdInfo can't access the register with card size.  this is the first access to the card so you may have a wiring problem.

What board are you using?  What type SD module and card?  How is it connected?

pmdmartins123, 2017-06-01:
Hi,

So I am  using an micro sd card reader bought from this store:
https://www.dfrobot.com/product-875.html
the sd card is HC 1 from Scan Disk.
The default wiring is:
13->SCK
12->MISO
11->MOSI
4->Chip Silect
But i have another stuff connected to this ports. So what I had to do was I changed the wiring to another ports and the code that i uploaded to the arduino was this:

#include <SPI.h>
#include <SdFat.h>

/**************SD Card Congiguration*****************/
#if ENABLE_SOFTWARE_SPI_CLASS
const uint8_t SOFT_MISO_PIN = 7;
const uint8_t SOFT_MOSI_PIN = 6;
const uint8_t SOFT_SCK_PIN  = 5;
const uint8_t SD_CHIP_SELECT_PIN = 4;
#else  // ENABLE_SOFTWARE_SPI_CLASS
#error ENABLE_SOFTWARE_SPI_CLASS must be set non-zero in SdFat/SdFatConfig.h
#endif  //ENABLE_SOFTWARE_SPI_CLASS

// SdFat software SPI template
SdFatSoftSpi<SOFT_MISO_PIN, SOFT_MOSI_PIN, SOFT_SCK_PIN> sd; // this ables to attribute aonther ports different of the default ports.

// Test file.
SdFile file;
SdFile hour;
File readFile;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  SPI.begin();
}

void loop(){
....
}

I hope this helps

pmdmartins123, 2017-06-01:
If you want to connect to the default wiring all you have to do is:

#include <SPI.h>
//#include <SD.h>
#include "SdFat.h"
SdFat SD; //default wiring

#define SD_CS_PIN SS
File myFile;

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.print("Initializing SD card...");

  if (!SD.begin(SD_CS_PIN)) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");

  // open the file. note that only one file can be open at a time,
  // so you have to close this one before opening another.
  myFile = SD.open("test.txt", FILE_WRITE);

  // if the file opened okay, write to it:
  if (myFile) {
    Serial.print("Writing to test.txt...");
    myFile.println("testing 1, 2, 3.");
    // close the file:
    myFile.close();
    Serial.println("done.");
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }

  // re-open the file for reading:
  myFile = SD.open("test.txt");
  if (myFile) {
    Serial.println("test.txt:");

    // read from the file until there's nothing else in it:
    while (myFile.available()) {
      Serial.write(myFile.read());
    }
    // close the file:
    myFile.close();
  } else {
    // if the file didn't open, print an error:
    Serial.println("error opening test.txt");
  }
}

void loop() {
  // nothing happens after setup
}

this code is ReadWrite.ino example.

greiman, 2017-06-01:
I can't help since I don't have your SD module.  The module uses resistor voltage dividers instead of IC level shifters.  This type SD module often causes problems.

pmdmartins123, 2017-06-01:
ok thanks

schlaegerz, 2017-06-01:
I actually know mine is wired correctly because it was working with the built-in SD card libraries. 

It turns out that it was a clock speed problem. When I changed it down to 1/2 clock speed everything seems to work just fine.

greiman, 2017-07-04:
built-in SD library use 4 MHz by default.  You can use higher clock speed with this undocumented call.

`  boolean begin(uint32_t clock, uint8_t csPin);`

closed 2017-07-04 by greiman