greiman/SdFat #59: Slow reading of free space

andreydung, 2017-06-10:
I'm checking free space of SD card (Sandisk, 64G, formatted by official SD formatter):

``` 
 uint32_t freeSpace = SD.vol()->freeClusterCount() * SD.vol()->blocksPerCluster();
```

This line takes 54s to finish. Is this normal? I'm using RFduino with SDfat (SPI of RFduino has been rewritten to work with SdFat).

stevstrong, 2017-06-10:
Running the standard SdInfo sketch will also take some time till "freeClusters: " display the result.

greiman, 2017-06-10:
beginTransaction() should set only the SPI speed , bit order, and data mode.   begin() will set the default speed/mode but may be slower since the SPI controller will be initialized. 

freeClusterCount() must read the entire FAT to find the free cluster count.  This will take a long time for a 64G card.  

You can limit reading the FAT to the first call by setting this symbol at about line 80 of SdFatConfig.h.

```
//------------------------------------------------------------------------------
/** 
 * Set MAINTAIN_FREE_CLUSTER_COUNT nonzero to keep the count of free clusters
 * updated.  This will increase the speed of the freeClusterCount() call
 * after the first call.  Extra flash will be required.
 */
#define MAINTAIN_FREE_CLUSTER_COUNT 0
```

stevstrong, 2017-06-11:
I tried to set `MAINTAIN_FREE_CLUSTER_COUNT to "1"` but it did not help, the SdInfo sketch still take the same amount of time to return freeClusterCount(). So I guess this is inevitable for the first time this function is called.

andreydung, 2017-06-11:
Thanks all! I tried a smaller card (32Gb) and with the new method it's down to 3 sec.

I just need to monitor and check that data collection to SD card is working fine. Are there any other alternative statistics that could be performed much faster? I'm looking for the range of 10ms.

greiman, 2017-06-11:
The first call to freeClusterCount() after SD.begin() will take a long time.

If MAINTAIN_FREE_CLUSTER_COUNT is nonzero, the time should be much less than a ms after the first call.

Here is the code:

```
int32_t FatVolume::freeClusterCount() {
#if MAINTAIN_FREE_CLUSTER_COUNT
  if (m_freeClusterCount >= 0) {
    return m_freeClusterCount;
  }
#endif  // MAINTAIN_FREE_CLUSTER_COUNT
```

closed 2017-07-04 by greiman