greiman/SdFat #60: 'ostream& endl(ostream&)' redeclared as different kind of symbol

paulalting, 2017-06-30:
Another Eclipse user here with SAM3X8E (DUE).
An issue I struck when including an often used library, streaming.h, which defines endl, as does iostream as part of SDFat.

Streaming library: http://arduiniana.org/libraries/streaming

Previous declaration of '_EndLineCode endl' error occurs at line Streaming.h line: 100
`enum _EndLineCode { endl };`

and iostream at line: 40
`inline ostream& endl(ostream& os) {`

To get error, simply include streaming.h with SdFat.h.

Any thoughts, I guess there would be a few users of Streaming.h by Mikal Hart.

greiman, 2017-06-30:
You can comment out the endl function in iostream.h.

paulalting, 2017-07-01:
Thanks Bill, that's what I did, had SD functionality working good.
Many thanks for the library.
Are there any consequences in terms of SdFat ?

greiman, 2017-07-01:
Many examples that use endl will not compile.

greiman, 2017-07-04:
close

closed 2017-07-04 by greiman