greiman/SdFat #62: Problem Importing SdFat

antonays, 2017-07-05:
Hi,

I am trying to import the library to my Arduino Sketch, using Arduino IDE 1.8.3 and Ubuntu 14.04.
I've downloaded the ZIP file and extracted its contents to my "local" sketchbook\libraries.

From the Arduino Library Manager, it seems that the library is recognized and shows up as "Installed".

However, trying to import the library into my project with
`#include "SdFat.h"
SdFat my_sd;`

I get the following error:
`test:12: error: 'SdFat' does not name a type
 SdFat my_sd;
 ^
exit status 1
'SdFat' does not name a type`

I have tried several fixes found online:
* i've tried extracting the zip both to the directory SdFat_master and renaming it to SdFat, no luck.

* tried with `extern SdFat my_sd`, no luck.

* tried to copy the contents of the /src directory directly to the SdFat directory, such that SdFat.h is directly under the library, also no luck.

Could this be an Ubuntu thing? Other external libraries seem to work fine, including the RTOSlogger from the same developer. At this point only SdFat is giving me this type of problems.

greiman, 2017-07-05:
I have Ubuntu 16.04 LTS on a machine that had no Arduino IDE installed.

I installed IDE 1.8.3 using [these](https://www.arduino.cc/en/Guide/Linux) instructions.

I then used the Library Manager to install SdFat Version 1.0.1.  I was able to compile and run several examples.

I just tagged version SdFat 1.0.3 so you may get that version soon with the library manager.  Arduino runs a job that scans github every hour.

I can't help with problems when you manually install a library on Linux.

antonays, 2017-07-06:
Thank you @greiman for the response.

I ended up discovering that the problem was in the way i imported the library. Apparently, in the case of SdFat, it's not enough to just copy the library to the libraries directory, but it is essential to actually use the Library Manager to import (Sketch->Include Library->Contributed Libraries->SdFat in Arduino IDE 1.8.3). Doing so, the IDE automatically generates the following imports
`#include <FreeStack.h>`
`#include <MinimumSerial.h>`
`#include <SysCall.h>`
`#include <BlockDriver.h>`
`#include <SdFat.h>`
`#include <SdFatConfig.h>`
Apparently it is critical to import them in THIS order and not change it.

Anyway, the issue is solved so i close it.

closed 2017-07-06 by antonays

greiman, 2017-07-06:
Look at the examples folder.  For most programs you only need this include.
`#include "SdFat.h"`

Several examples use FreeStack to determine memory use. FreeStack has no SdFat code and requires
#include "FreeStack.h"

Finally, MinimumSerial.h has no SdFat code and is only required if you want to use it as a replacement for Serial on small AVR boards like UNO.

antonays, 2017-07-06:
@greiman I am using Arduino Nano, so maybe the MinimumSerial.h is related to this.

Anyway, these 6 imports were auto-generated by the IDE when importing SdFat.

greiman, 2017-07-06:
No, MinimumSerial is not used in any SdFat code.  It is only used in the LowLatencyLogger examples to save about 250 bytes of RAM on 328 boards.

greiman, 2017-07-06:
The IDE just adds all the .h files in the src directory.  On windows they are in alpha order like this:
```
#include <BlockDriver.h>
#include <FreeStack.h>
#include <MinimumSerial.h>
#include <SdFat.h>
#include <SdFatConfig.h>
#include <SysCall.h>
```
On Linux the order seems random.   On the Linux machine I used for a test I get this order:
```
#include <FreeStack.h>
#include <BlockDriver.h>
#include <SdFat.h>
#include <SdFatConfig.h>
#include <MinimumSerial.h>
#include <SysCall.h>
```

RemyDavies, 2024-01-01:
I'm experiencing the same thing with Arduino IDE 2.2.1

I have downloaded the library via the library manager.

Then, I add the library to my project:
(Sketch->Include Library->Contributed Libraries->SdFat)

It then adds all of the following includes:
#include <BufferedPrint.h>
#include <FreeStack.h>
#include <MinimumSerial.h>
#include <RingBuf.h>
#include <SdFat.h>
#include <SdFatConfig.h>
#include <sdios.h>

What is going on here?

greiman, 2024-01-01:
> What is going on here?

The IDE thinks all include files in the `src` directory need to be added for a library.

You probably only need to include SdFat.h.