greiman/SdFat #63: Error codes 0X20,0XFF and 0

joaohenriquef, 2017-08-01:


greiman, 2017-08-01:
Looks like you are using an ESP8266 which has a fast SPI clock.

My guess is that you have marginal wiring.  Try using a very slow SPI speed by setting the second argument of begin to one of these values, SPI_QUARTER_SPEED, SPI_EIGHTH_SPEED.
```
  if (!sd.begin(csPin, SPI_QUARTER_SPEED)) {
    // Handle errror.
 }
```
The error code are an enum in SdFat/src/SdCardSdInfo.h

> 20 is SD_CARD_ERROR_CMD0 - Card did not respond to the reset command.
> 43 is SD_CARD_ERROR_ACMD41 - Card did not respond to "Activate card initialization".
> 50 is SD_CARD_ERROR_READ - Read error.

This means you are getting errors on most of the initialization commands.

joaohenriquef, 2017-08-02:
Thank you very much for your help.

Unfortunatelly I still get the same errors using SPI_QUARTER_SPEED or SPI_EIGHTH_SPEED.

Any other guesses?

greiman, 2017-08-02:
I can't help you.  I tested the current SdFat a few minutes ago with an  Adafruit HUZZAH ESP8266 Breakout.

I used [this ](http://www.ebay.com/itm/2-Pcs-TF-Mini-Micro-SD-Card-Memory-Module-For-Arduino-ARM-AVR-/361974001960?hash=item544754f128:g:nAkAAOSwcgNZD-JK) sd module.

I uses a 16GB SanDisk card and ran SdInfo with no mods.

> 
> type any character to start
> 
> init time: 117 ms
> 
> Card type: SDHC
> 
> Manufacturer ID: 0X3
> OEM ID: SD
> Product: SE16G
> Version: 8.0
> Serial number: 0X91D1262A
> Manufacturing date: 10/2015
> 
> cardSize: 15931.54 MB (MB = 1,000,000 bytes)
> flashEraseSize: 128 blocks
> eraseSingleBlock: true
> OCR: 0XC0FF8000
> 
> SD Partition Table
> part,boot,type,start,length
> 1,0X0,0XC,8192,31108096
> 2,0X0,0X0,0,0
> 3,0X0,0X0,0,0
> 4,0X0,0X0,0,0
> 
> Volume is FAT32
> blocksPerCluster: 64
> clusterCount: 485936
> freeClusters: 485778
> freeSpace: 15917.97 MB (MB = 1,000,000 bytes)
> fatStartBlock: 8790
> fatCount: 2
> blocksPerFat: 3797
> rootDirStart: 2
> dataStartBlock: 16384

joaohenriquef, 2017-08-11:
I see. I'm still trying to solve this, when I do some progress I'll update here. Thanks.

eddstep, 2017-08-14:
the same problem. with SD library get card info , but here is `Assuming the SD chip select pin is: 15
Edit SD_CHIP_SELECT to change the SD chip select pin.

type any character to start

init time: 4 ms

Card type: SDXC
error: readCID failed
SD errorCode: 0X53,0XFF
`

greiman, 2017-08-14:
Can't help. I have no Idea what your hardware setup is.

closed 2017-08-14 by greiman

joaohenriquef, 2017-08-15:
It was a simple mistake. Vcc was connected to 3.3V but the SD module only works with 5V.

tiseidel, 2020-08-12:
> 
> 
> It was a simple mistake. Vcc was connected to 3.3V but my SD module only works with 5V.

If anyone else stumbles across this aswell: 
Changing Vcc to 5V and setting the speed to:

`
 SDCard.begin(CS, SD_SCK_MHZ(10)))
`

did the trick for me (ESP32).