greiman/SdFat #64: No method to force un-cached reads in the FatFile class

pauley-unsaturated, 2017-08-06:
It looks like there is no way to force all reads to be un-cached (and avoid an unnecessary buffer / copy) in File and the underlying FatFile classes.

Specifically: at FatFile.cpp:750
` if (offset != 0 || toRead < 512 || block == m_vol->cacheBlockNumber()) {`

will always happen for the last block in a file if it's not a full block. This is because toRead is clipped at the file size. 

I am doing my own buffering above this layer, and I am always reading in multiples of the block size, but it looks like if my file is not an even multiple of 512 bytes, I will always engage the volume cache at the end.

pauley-unsaturated, 2017-08-06:
Here is my ghetto hack to attain my desired behavior:

```
diff --git a/src/FatLib/FatFile.cpp b/src/FatLib/FatFile.cpp
index 8f3a820..af62ea2 100644
--- a/src/FatLib/FatFile.cpp
+++ b/src/FatLib/FatFile.cpp
@@ -701,6 +701,7 @@ int FatFile::read(void* buf, size_t nbyte) {
   size_t toRead;
   uint32_t block;  // raw device block number
   cache_t* pc;
+  bool noCacheClipped = false;
 
   // error if not open for read
   if (!isOpen() || !(m_flags & O_READ)) {
@@ -711,6 +712,9 @@ int FatFile::read(void* buf, size_t nbyte) {
   if (isFile()) {
     uint32_t tmp32 = m_fileSize - m_curPosition;
     if (nbyte >= tmp32) {
+      if (!(nbyte & 0x1FF)) {
+        noCacheClipped = true;
+      }
       nbyte = tmp32;
     }
   } else if (isRootFixed()) {
@@ -750,7 +754,9 @@ int FatFile::read(void* buf, size_t nbyte) {
       }
       block = m_vol->clusterFirstBlock(m_curCluster) + blockOfCluster;
     }
-    if (offset != 0 || toRead < 512 || block == m_vol->cacheBlockNumber()) {
+    if (offset != 0
+        || (toRead < 512 && !noCacheClipped)
+        || block == m_vol->cacheBlockNumber()) {
       // amount to be read from current block
       n = 512 - offset;
       if (n > toRead) {
@@ -789,7 +795,12 @@ int FatFile::read(void* buf, size_t nbyte) {
 #endif  // USE_MULTI_BLOCK_IO
     } else {
       // read single block
-      n = 512;
+      if (toRead < 512 && noCacheClipped) {
+        n = toRead;
+      }
+      else {
+        n = 512;
+      }
       if (!m_vol->readBlock(block, dst)) {
         DBG_FAIL_MACRO;
         goto fail;

```

greiman, 2017-08-07:
You can't skip the cache just because you are reading 512 bytes.  Reads must also start on 512 byte boundaries.

The only way to avoid the  cache is to read multiples of 512 bytes only so each read starts on a file position that is a multiple of 512 bytes.

closed 2017-08-07 by greiman

greiman, 2017-08-07:
I am not about to return one value for the number of bytes read but change bytes in a buffer beyond that.

pauley-unsaturated, 2017-08-08:
Yeah, it’s inelegant. It turns out my performance issue was due to SPI frequency, thanks for considering this, and thanks for the library!