greiman/SdFat #65: Teensey 3.6 directory with openNext returning false

SimpleThoughts, 2017-08-25:
Awesome project! Kudos.

When using a Teensey 3.6 to list directory the openNext always returns false.  Ideas? Example? 

SdFatSdio sd;

...
...   Initializes and reads files perfectly if opened by name.
...   Pulled the initialize and directory from demo code

  SdFile file;
  SdFile dirFile;

  if (!dirFile.open("/", O_READ)) {
	  ... display error ...
  }
  while (file.openNext(&dirFile, O_READ)) {
	  // Skip directories and hidden files.
	  if (!file.isSubDir() && !file.isHidden()) {
		  char szName[255];
		  file.getName(szName, 255);
		  ... use ... 
	  }
	  file.close();
  }
...

***Please pardon my ignorance/stupidity if I'm missing something obvious (which I probably am). Thank you for all the hard work.  😃

greiman, 2017-08-25:
I don't see the problem.

I Tried this on a Tensy 3.6.

```
/*
   Print size, modify date/time, and name for all files in root.
*/
#include "SdFat.h"
SdFatSdio sd;
SdFile file;
SdFile dirFile;
//------------------------------------------------------------------------------
void setup() {
  Serial.begin(9600);

  // Wait for USB Serial
  while (!Serial) {
    SysCall::yield();
  }
  Serial.println("Type any character to start");
  while (!Serial.available()) {
    SysCall::yield();
  }
  if (!sd.begin()) {
    sd.initErrorHalt("SdFatSdio begin() failed");
  }
  if (!dirFile.open("/", O_READ)) {
    sd.errorHalt("dirFile.open failed");
  }
  while (file.openNext(&dirFile, O_READ)) {
    if (!file.isSubDir() && !file.isHidden()) {
      file.printFileSize(&Serial);
      Serial.write(' ');
      file.printModifyDateTime(&Serial);
      Serial.write(' ');
      file.printName(&Serial);
      if (file.isDir()) {
        // Indicate a directory.
        Serial.write('/');
      }
      Serial.println();
    }
    file.close();
  }
  Serial.println("Done!");
}
//------------------------------------------------------------------------------
void loop() {}
```
Here is the output.
```
Type any character to start
   8388608 1980-00-00 00:00:00 TeensyDemo.bin
  49987584 1980-00-00 00:00:00 bench.dat
Done!
```

If I remove the if (!file.isSubDir() && !file.isHidden()) test I get this 
```
Type any character to start
   8388608 1980-00-00 00:00:00 TeensyDemo.bin
  49987584 1980-00-00 00:00:00 bench.dat
         0 1980-00-00 00:00:00 Dir1/
         0 1980-00-00 00:00:00 Dir2/
         0 2017-08-25 09:42:18 System Volume Information/
Done!
```

SimpleThoughts, 2017-08-26:
Bill,
   Well, it is official, I am an idiot. 

   Made a new project and pasted in your code. Nothing. I updated every library. Nothing worked. Tried three different SD cards.  Again, nothing.

   That’s when, in desperation, my brain started working! All three SD cards were originally from Raspberry Pi kits. I re-tasked the 8GB cards when they were upgraded.

   The old Linux partition is readable by the default SD libraries, Windows, and of course Linux. The partition, however, isn’t correct. Removed the bad partitions and re-formatted and bingo!

Awesome, awesome, work!

closed 2017-08-26 by SimpleThoughts