greiman/SdFat #66: yield has not been declared

physicien, 2017-09-02:
Hi, whenever I try to compile an example include with the library, I have the same error.

```
In file included from /usr/share/arduino/libraries/SdFat/SdFat.h:26:0,
                 from bench.ino:5:
/usr/share/arduino/libraries/SdFat/SysCall.h: In static member function ‘static void SysCall::yield()’:
/usr/share/arduino/libraries/SdFat/SysCall.h:74:3: error: ‘::yield’ has not been declared
   ::yield();
   ^
In file included from /usr/share/arduino/libraries/SdFat/FatLib/FatLib.h:23:0,
                 from /usr/share/arduino/libraries/SdFat/SdFat.h:28,
                 from bench.ino:5:
/usr/share/arduino/libraries/SdFat/FatLib/ArduinoStream.h: In member function ‘void ArduinoInStream::readline()’:
/usr/share/arduino/libraries/SdFat/FatLib/ArduinoStream.h:53:13: error: ‘yield’ was not declared in this scope
       yield();
```

Any idea?

greiman, 2017-09-03:
I have no idea since you didn't give details about the hardware or software you are using.

[yield(](https://www.arduino.cc/en/Reference/SchedulerYield)) is a standard function in the Arduino IDE.

physicien, 2017-09-03:
You're right, the problem was my version of Arduino IDE. Debian is too stable sometimes.

closed 2017-09-03 by physicien

fgfdgddfg, 2019-03-19:
How you solved this ploblem on debian?