greiman/SdFat #71: GPL v3 license is a poison pill for an approved medical device

cfeied, 2017-09-30:
Hi,

I'm a physician who has developed a Teensy-based medical device, and I have learned from companies interested in the device that the use of GPL v3 libraries is an "absolute poison pill."   This is because approved medical devices cannot permit source code to be modified and reloaded, nor can they permit dynamic linking to incorporate new versions of libraries. FDA approval requires that the device forever operate exactly as released, and any code change is a big issue. Thus, static linking only.

My understanding from Wikipedia and GNU.org that the **_intended_** effect of GPLv3 is generally to prevent proprietary commercial developers from using a library, thus giving an advantage to open-source developers. As a philosophical stance I understand this. I've contributed my share of open source software, and although I personally think the MIT license creates the greatest overall utility, I get it that some people feel otherwise.

So I'm not proposing anything, I'm just trying to make sure that I understand correctly: to your knowledge and by your intent, there is **_no legal mechanism_** by which I can use SdFat (or any library that invokes or links it) in an embedded device (such as a medical device) for which it will not be possible to release source code or enable dynamic linking.

Is that correct?

Thanks,

-- Craig

greiman, 2017-10-01:
Yes, GPLv3 is very restrictive.  I have been developing a replacement for SdFat that also supports exFAT and I am licensing it with the more permissive MIT license.  It is the SdFs archive here on GitHub.

I plan to change SdFat to MIT along with other repositories.  I will try to find time to edit all the SdFat files.  I can automate it a bit.

cfeied, 2017-10-03:
That will be awesome!

greiman, 2017-10-04:
I changed to MIT license.  

Thanks for bumping me, I had been putting off doing the edits by finding other more fun things to do first.

paqwa, 2017-10-04:
Wow, that's great, is there a link to your SDFs covered under MIT license or is it still in early development? Great work as usually. Thanks.

greiman, 2017-10-05:
[SdFs](https://github.com/greiman/SdFs)

paqwa, 2017-10-05:
Thank you very much, I'll try it out over the weekend if I get a chance.

paqwa, 2017-10-17:
Hi Bill, 
I was trying to use your SdFs library with a Teensy 3.6 using the built in sd card. To keep things simple I tried to start with the examples provided with the library, so I started by using the 'TeensySdioDemo.ino' sketch. However I keep getting an error upon compilation and am unsure of where to look to fix it. The examples look to have been edited and commented for this new version so I'm sure I'm doing something silly. The error as reported in the verbose output of the Arduino IDE is as follows (from version 1.8.1). Any idea what's going wrong, I'm using the example completely unedited as I believe it's already setup for the Teensy 3.6. Any advice would be much appreciated. Thanks.

> In file included from C:\SDFs_Test\libraries\SdFs-master\src/SdCard/SdSpiCard.h:35:0,

                 from C:\SDFs_Test\libraries\SdFs-master\src/SdCard/SdCard.h:28,

                 from C:\SDFs_Test\libraries\SdFs-master\src/SdFs.h:32,

                 from C:\SDFs_Test\TeensySdioDemo\TeensySdioDemo.ino:4:

c:\sdfs_test\libraries\sdfs-master\src\spidriver\sdspidriver.h: In member function 'void SdSpiLibDriver::begin(SdSpiConfig)':

c:\sdfs_test\libraries\sdfs-master\src\spidriver\sdspidriver.h:123:13: error: cannot convert 'SPI1Class*' to 'SPIClass*' in assignment

       m_spi = &SDCARD_SPI;

             ^

Using library SdFs-master at version 2017.8.14 in folder: C:\SDFs_Test\libraries\SdFs-master 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SPI 
Error compiling for board Teensy 3.6.

paqwa, 2017-10-17:
Hi Bill, 
As a minor update to my comment directly above, I found that if I comment out one line in your library (see below), the TeensySdioDemo.ino sketch will compile and run on the Teensy 3.6. The Teensy 3.6 built-in SD slot has 4 byte FIFO capabilities which I had though was executed via DMA but I'm only learning about such things so please excuse my ignorance if I'm mistaken. If the built in SD card can run even faster than it does with the 4 byte FIFO buffer I'd be very interested. 

This is the line I commented out
#define SDCARD_SPI      SPI1  (line number 200 from file FsConfig.h of your SdFs library) 

After running the compiled sketch, the user option Type 1 for FIFO SDIO works but option 2 for DMA SDIO locks up the device. However, with FIFO SDIO it is giving me about the same speed as I was getting with your SdFAT library using class SdFatSdioEX. Results shown below. Do you think it's possible to get this running even faster using the builtin SD on Teensy 3.6? It is pretty good as it is. Also, would you advise that I take this approach in commenting out this option to force your SdFs library to use the standard SPI class before I use in a larger project?  Thanks for sharing the library and for any advise on this matter. 

Results from running TeensySdioDemo shown here:

> Type '1' for FIFO SDIO
     '2' for DMA SDIO
     '3' for Dedicated SPI
     '4' for Shared SPI

FIFO SDIO mode.

size,write,read
bytes,KB/sec,KB/sec
512,17120.38,19228.86
1024,16740.55,19489.26
2048,15018.81,19768.09
4096,17965.64,19907.61
8192,16868.44,19972.59
16384,17035.99,20014.19
32768,18063.44,20030.49

totalMicros  6440968
yieldMicros  334006
yieldCalls   183
yieldMaxUsec 6857
Done

greiman, 2017-10-17:
You need to update your version of Arduino/Teensyduino to get rid of the compile error.

I am using Arduino IDE version 1.8.5 with Teensyduino 1.40.

The SDIO with FIFO access is fastest.   The SDIO controller has a 512 byte FIFO.

The SDIO controller has errata that prevent efficient multiple small transfers using DMA.  DMA uses less CPU so some applications use DMA with large, 32KB writes.

The speeds you are seeing are about the limit for the Teensy SDIO bus.  Teensy has a 45 MHz 4-bit SDIO bus so the limit is 22.5 MB/sec.

If you ever access an SD card in SPI mode, you must power cycle the card to return to SDIO mode.  You should be able to use both FIFO and DMA SDIO without locking the card.

The Teensy SPI controller has a small four deep FIFO.  I think the latest version of Teensy uses DMA with SPI.  Dedicated SPI with the built-in SD will be much slower since it is a one-bit bus at 30 MHz.

paqwa, 2017-10-17:
This is very informative, thank you very much. I'll update my Arduino IDE and Teensyduino Versions as suggested and I expect all with work for me too. Thanks again. Pat

closed 2018-01-22 by greiman