greiman/SdFat #72: Odd behavior from ADC

eelife, 2017-09-30:
Hey all, I am using the sdfatlib library to record ADC input to eventually read strain gauge data. At the moment I just have it connected to a 1-2 voltage divider. The rail voltage on my MCU (adafruit featherboard) is 3.3v so i am expecting to see ~512 from the 10bit adc output. Which I do see for the beginning of my measurements. However, as I keep recording, the values decrease slowly until they reach 0... Is this an issue with my read speed. Here is the current code. https://github.com/greiman/SdFat/tree/master/examples/AnalogBinLogger. I made slight changes to the software to make it compatible with my board. For instance I am running ADC prescaler 5 which is CPU/32.
Here is the csv of the data. http://docdro.id/zUAUJnu
Is the issue that I am running the ADC to fast, and it can't keep up? Using generic analogReads on A0, i am getting 518-519, which is the expected value. What is the difference between analogRead and what I am using here?

greiman, 2017-10-01:
What are the resistance values of the voltage divider?  What is your sample rate?

F_CPU/32 should be OK.  I believe the 3.3V adafruit featherboard runs at 8 Mhz.

The AVR ADC drags down high impedance sources at high sample rates.

Each time the ADC samples a source, it charges a sample and hold cap. 

Try measuring a 1.5 V battery connected directly to the ADC.  If it measures correctly, it's an impedance problem.

If it is an impedance problem, you may need an op-amp buffer.

Let me know what happens.

eelife, 2017-10-01:
Will do, no access to it today, but I will let you know what happens tomorrow. It could be an impedance issue as I was using 2 10khom resistors for the voltage divider (didn't even think doh...). But it was working perfectly on the simple analogRead, which was throwing me off. What does analogRead do differently than what you are doing in your library, obviously excluding the SD card stuff?

You are right the featherboard is 3.3V at 8Mhz.

Thanks for the help!

greiman, 2017-10-01:
> What does analogRead do differently than what you are doing in your library, obviously excluding the SD card stuff?

The main thing is more samples per second.

I am beginning to suspect some other problem.  It's like the ADC mux is not being set to your pin.

Have you set PIN_LIST like this for pin A0?
`const uint8_t PIN_LIST[] = {0};`

I suspect you will have the same problem after the battery test

eelife, 2017-10-01:
Yes, I also suspect an issue with the MUX, but I have set the PIN_LIST[] = {0} exactly like that.

greiman, 2017-10-01:
I ran AnalogBinLogger on an Uno with A0 connected to 3.3V, A1 connected to 5V and A3 connected to gnd.

The only mod was:
`const uint8_t PIN_LIST[] = {0};`

I get mostly 690 and 689s an few 691s.

The Uno is connected to a USB and the 5V pin measures 4.933 and the 3.3V pin measures 3.311.

4.933*689.5/1023 = 3.325.  So it's as close as you can expect.

eelife, 2017-10-01:
why a3 connected to gnd?

eelife, 2017-10-02:
Figured it out. I had to scour through the datasheet. The featherboard atmega32u4 (MCU on board), does not have a 3.3V reference, which is odd... but instead uses a 2.56 internal reference by setting REFS1 and REFS0 both to 1. Thanks for the help!

greiman, 2017-10-03:
I see you solved the problem.  I put other voltage on several pins during the test to make sure I am really measuring the correct pin and that the other pins are not interfering.

eelife, 2017-10-03:
I get constant values, but I am still having issues with getting the full resolution.

greiman, 2017-10-04:
What do you mean by full resolution?

You might look at [extras/AnalogBinLogger](https://github.com/greiman/SdFat/tree/master/extras/AnalogBinLoggerExtras).