greiman/SdFat #73: i cant find the file:Sd2PinMap.h to change in SDfatlib!

hadid5, 2017-10-01:
hey guys,

im trying this example code but i can't get it applied in arduino.
i need to change this file name Sd2PinMap.h in > SDfatlib.

could someone help me out?

thanks!!

greiman, 2017-10-01:
Which example? 

Which Arduino board are you using?

What version of the Arduino IDE are you using?

Any other helpful information?

hadid5, 2017-10-02:
Thanks for the respond!
I am using the arduino uno version 65139.
I used the mp3 example code on this link.
https://www.floris.cc/shop/en/shields/166-mp3-player-shield.html
I used the file mp3 player controle example.

If u need Any more info let me know. Im happy u want to help me out here!

hadid5, 2017-10-03:
anyone who could help me out here?

greiman, 2017-10-03:
Contact the author of the mp3 example.

closed 2017-10-03 by greiman

hadid5, 2017-10-03:
they give me the link to your gitbub.,....

greiman, 2017-10-03:
The version of SdFat used in the mp3 examples is no longer supported.  I can't fix other authors old code.
```
 4-28-2011
 Spark Fun Electronics 2011
 Nathan Seidle
```