greiman/SdFat #75: suddenly cant open any file after days of proper operation.

geetee24, 2017-10-21:
Is tjhere any way to get the reason why a file cannot be opened anymore sdfile.open()..  The card IS 100% fine.

Many thanks.

greiman, 2017-10-21:
How do you know the card is fine?

The Arduino uses SPI mode, PCs and other devices use SDIO mode so testing on a PC/Mac is not proof.

Have you checked the card for corruption? 

Backup files before using repair programs.

On Windows use File Explorer or chkdsk.  Use fsck or similar program on Mac or Linux.

Restore the standard SD format with [SD Formatter](https://www.sdcard.org/downloads/formatter_4/). 

Don't use an OS utility to format SD cards, OS utilities often do not produce the format the card was designed to use.

geetee24, 2017-10-21:
card is fine.  already did all that you had suggested.   how do i debug the fact that i cant open the file from the arduino.  is there an error code that i can display upon the failed open?

geetee24, 2017-10-21:
this AM i turned on my arduino (with no code change from night before) and it again works 10% fine.   again, is there a error code that I can read to explain why the weird behavior?

Thanks

greiman, 2017-10-21:
>  already did all that you had suggested

How did you test that the card's SPI interface is OK?  You must test on another SPI based system.

You can print any hardware error code after open fails by calling:

` sd.errorPrint(&Serial);`

It will print something like this if there is a hardware problem.

> SD errorCode: 0X64,0X0

The error codes are defined in SdFat/src/SdCard/SdInfo.h.

There are many causes of open failures, a common cause is to little free memory.  Another is a bug that writes over SdFat globals.  More than 95% of problems like yours are in this category or due to marginal wiring. 

Try some SdFat Examples.  If  examples like OpenNext, SdInfo, and bench work, you most likely have a problem in your program.

geetee24, 2017-10-21:
thanks.  now it is failing again.  and sdFat.errorPrint ( &Serial );

printing NOTHING.

Why no error returned?

geetee24, 2017-10-21:
what are SdFat globals?

geetee24, 2017-10-21:
I have 2,000 bytes free ram  when I do a file open

greiman, 2017-10-21:
> thanks. now it is failing again. and sdFat.errorPrint ( &Serial );
> 
> printing NOTHING.

No hardware error.  possible a bad file name.

You didn't run the examples.  Do you want help?

geetee24, 2017-10-24:
loop ()

	getRequestFromBrowser {
	
		open file
		send file out
		close file
		
		return
		
		
if index.html requests 5 files from arduino, the above code (sdfat fails to open files those 5 files).

if i manually ask the arduino to open each of the 5 files, one at a time (slowly), each is returned just perfectly.

Question:  Is their anything in SdFat that could explain this issue?

Thanks

greiman, 2017-10-25:
> if index.html requests 5 files from arduino, the above code (sdfat fails to open files those 5 files).

This is not code.  I can't guess what hardware you are using, what other libraries/software you are using and what the actual code does.

I don't have time to look at the dozen or more request for help like this that I get every day and debug user code.

If you find a problem with SdFat, post a simple example that illustrates the issue.

sap586, 2017-11-03:
suddenly can't create any new files
SD errorCode: 0X1,0XFF

greiman, 2017-11-03:
The first SD initial command is failing.  Could be wiring, wrong chip select ...

I can't help since I don't have your hardware.

Mr-Newlove, 2020-07-26:
I am using this lib in a small project, but we had 100 boards made after everything seems fine, and now this problem seems to start happening randomly on about 1 in 10 devices tested. Can happen immediately, or after months of running fine. Always with:

SD errorCode: 0X57,0XED

greiman, 2020-07-26:
SD errorCode: 0X57,0XED is a hardware write error.  The error occurs at line 739 of SdSpiCard.cpp.

The error may be due to poor signals on the SPI bus.  I can't help much since it is at the hardware level.

The error token contained in m_status only indicates the card can't program the flash block.  This could be due to a garbled start of data token as seen by the card.  The start of data token is sent at line 733.

I would first look at the SPI lines to make sure the signals are clean.  If you have other SPI devices, make sure they are not interfering.

I don't have much luck helping with hardware debug when I can't play with the hardware.  I have scopes, logic analyzers and test setups and these give me clues that users don't see.

Mr-Newlove, 2020-07-27:
That is strange because I have one unit here that does it consistently after about 5 min of running.

I was looking at the lines with my scope but it was nice and sharp. I eventually narrowed it down to happening when trying to delete a lock flag file I am saving and then trying to open another right after.

Essentially I changed a line that was...
`SD_card.remove(buf);`
to...
`if (SD_card.exists(buf)) SD_card.remove(buf);`

Where I was not checking if the file actually exists or not. And then things are working fine again on these limited devices with problems.