greiman/SdFat #77: RAM usage

andrew-davie, 2017-10-24:
I am successfully using this fantastic library in my mechanical televisor (see https://www.youtube.com/watch?v=LIY_QkTzIX4 for a short video).  It's based on an Arduino Mega, and I'm so desperately short of RAM (I need about 128 bytes from somewhere), I'm thinking of delving into the SDFat library to see what I can scavenge. I understand there is a 512 byte buffer used by the library, and I'm wondering if it might be possible to reduce that (to, say, 384) and what effects that might have. Thought I'd ask here first....

greiman, 2017-10-24:
You can not reduce the size of the 512 byte sector buffer.

closed 2017-10-24 by greiman

andrew-davie, 2017-10-24:
Thanks for the quick reply.
Cheers
A