greiman/SdFat #78: Warnings with -O3-mpure-code  -flto (Teensy)

FrankBoesing, 2017-10-25:
"C:\Arduino\hardware\teensy/../tools/arm/bin/arm-none-eabi-gcc" -O3 -mpure-code -D__PURE_CODE__ -flto -fno-fat-lto-objects -Wl,--gc-sections,--relax,--defsym=__rtc_localtime=1508938828 "-TC:\Arduino\hardware\teensy\avr\cores\teensy3/mk66fx1m0.ld" -lstdc++ -fuse-linker-plugin -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -o "f:\arduino_build_642126/c64.ino.elf" "f:\arduino_build_642126\sketch\c64.ino.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\IntervalTimer.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\Teensy64.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\cia1.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\cia2.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\cpu.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\font_Play-Bold.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\font_Play-Regular.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\ili9341_t64.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\keyboard.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\output_dac.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\patches.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\pla.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\roms.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\usb_ehci.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\usb_enumeration.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\usb_hub.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\usb_keyboard.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\usb_memory.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\usb_print.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\util.cpp.o" "f:\arduino_build_642126\libraries\Teensy64_new\vic.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\MinimumSerial.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\FatLib\FatFile.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\FatLib\FatFileLFN.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\FatLib\FatFilePrint.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\FatLib\FatFileSFN.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\FatLib\FatVolume.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\FatLib\FmtNumber.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\FatLib\StdioStream.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\FatLib\fstream.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\FatLib\istream.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\FatLib\ostream.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\SdCard\SdSpiCard.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\SdCard\SdSpiCardEX.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\SdCard\SdioCardEX.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\SdCard\SdioTeensy.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\SpiDriver\SdSpiESP8266.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\SpiDriver\SdSpiSAM3X.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\SpiDriver\SdSpiSTM32F1.cpp.o" "f:\arduino_build_642126\libraries\SdFat-master\SpiDriver\SdSpiTeensy3.cpp.o" "f:\arduino_build_642126\libraries\SPI\SPI.cpp.o" "f:\arduino_build_642126\libraries\reSID\reSID.cpp.o" "f:\arduino_build_642126\libraries\reSID\sid.cpp.o" "f:\arduino_build_642126\libraries\uVGA-master\font8x8.cpp.o" "f:\arduino_build_642126\libraries\uVGA-master\uVGA.cpp.o" "f:\arduino_build_642126\libraries\uVGA-master\uVGA_DMA_RGB332.cpp.o" "f:\arduino_build_642126\libraries\uVGA-master\uVGA_gfx.cpp.o" "f:\arduino_build_642126/core\core.a" "-Lf:\arduino_build_642126" -larm_cortexM4lf_math -lm
**C:\Users\Frank\Documents\Arduino\libraries\SdFat-master\src/SdCard/SdSpiCard.h:37:7: warning: type 'struct SdSpiCard' violates one definition rule [-Wodr]**

 class SdSpiCard : public BaseBlockDriver {

       ^

C:\Users\Frank\Documents\Arduino\libraries\SdFat-master\src\SdCard\SdSpiCard.h:37:7: note: a different type is defined in another translation unit

 class SdSpiCard : public BaseBlockDriver {

       ^

C:\Users\Frank\Documents\Arduino\libraries\SdFat-master\src/SdCard/SdSpiCard.h:293:16: note: the first difference of corresponding definitions is field 'm_spiDriver'

   SdSpiDriver *m_spiDriver;

                ^

C:\Users\Frank\Documents\Arduino\libraries\SdFat-master\src\SdCard\SdSpiCard.h:293:16: note: a field of same name but different type is defined in another translation unit

   SdSpiDriver *m_spiDriver;

                ^

c:\users\frank\documents\arduino\libraries\sdfat-master\src\spidriver\sdspidriver.h:296:24: note: type name 'SdSpiLibDriver' should match type name 'SdSpiAltDriver'

 typedef SdFatSpiDriver SdSpiDriver;

                        ^

c:\users\frank\documents\arduino\libraries\sdfat-master\src\spidriver\sdspidriver.h:296:24: note: the incompatible type is defined here

 typedef SdFatSpiDriver SdSpiDriver;

                        ^

greiman, 2017-10-25:
I need more information.

I tried several SdFat examples with TeensyDuino 1.40 and Arduino IDE 1.8.5 and they compile and run.

I tried SdInfo with SPI and SDIO.
The bench example with SPI and SDIO.
The TeensySdioDemo
The ReadWrite demo.

greiman, 2017-10-25:
I suspect the two definitions occur due to a missing include of a config file.

Line numbers in latest SdFat do not match you output.

This is currently line 42 of SdSpiCard.h, not line 37.

`class SdSpiCard : public BaseBlockDriver {`

greiman, 2017-10-25:
I compile with this and don't get warnings.

-Wall  -Wodr -O3 -mpure-code -D__PURE_CODE__ -flto

FrankBoesing, 2017-10-26:
yes, works - thank you !
frank.

closed 2017-10-26 by FrankBoesing