greiman/SdFat #79: Sd card write error

tothtechnika, 2017-11-03:
Hi,

I tried to use the SD card library in Arduino with ESP8266 and used this code to test the write operation: 
```c
#include <SPI.h>
#include <SdFat.h>
#include "FreeStack.h"

const int chipSelect = D8;
SdFat sd;

void test_write()
{
  SdFile sdf;
  char buf[2048];
  uint16_t i;
  int16_t ret;
  
  for(i=0; i<2048; i++) buf[i]='a';

  ret = sdf.open("test_sd_file.txt", FILE_WRITE);
  sdf.seekSet(0);
  
  Serial.printf("\n\n\nopen: %d \n", ret);
  for(i=0; i<20; i++) {
    ret = sdf.write(buf, 2048);
    Serial.printf("write %d ret: %d \n", i, ret);
  }

    ret = sdf.write(buf, 1252);
    Serial.printf("write 1252 byte ret: %d \n", ret);
    
    sdf.close();  
}

void setup() {
  Serial.begin(115200);
  // put your setup code here, to run once:
  if (!sd.begin(chipSelect, SPI_HALF_SPEED)) {
    Serial.println("initialization failed!");
    return;
  }
  test_write();
}
```

And I saw this in the terminal:
```
open: 1 
write 0 ret: 2048 
write 1 ret: -1 
write 2 ret: -1 
write 3 ret: 2048 
write 4 ret: -1 
write 5 ret: -1 
write 6 ret: 2048 
write 7 ret: -1 
write 8 ret: -1 
write 9 ret: 2048 
write 10 ret: -1 
write 11 ret: -1 
write 12 ret: 2048 
write 13 ret: -1 
write 14 ret: -1 
write 15 ret: 2048 
write 16 ret: -1 
write 17 ret: -1 
write 18 ret: 2048 
write 19 ret: -1 
write 1252 byte ret: -1 
```
The file contains 14.336 bytes (7 * 2.048)

Adding some delay like the following results error only at the last (1252 byte) write:
```c
  for(i=0; i<20; i++) {
    ret = sdf.write(buf, 2048);
    Serial.printf("write %d ret: %d \n", i, ret);
    delay(1000);
  }
```

What could be the problem?
Thank you in advance!

greiman, 2017-11-03:
I can't help, I don't have your hardware.

closed 2017-11-03 by greiman

greiman, 2017-11-04:
I was able to run the bench example on an Adafruit ESP8266 with 2048 byte reads/writes.

> Type is FAT32
> Card size: 15.93 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X3
> OEM ID: SD
> Product: SE16G
> Version: 8.0
> Serial number: 0X91D1262A
> Manufacturing date: 10/2015
> 
> File size 5 MB
> Buffer size 2048 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 1285.46,76864,1390,1591
> 1213.68,188344,1429,1685
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 1393.30,1932,1406,1469
> 1391.75,1929,1385,1470
> 
> Done