greiman/SdFat #81: SD card write super slow after a not finsihed write

schlaegerz, 2017-11-10:
I am having trouble where if my ESP8266 shuts off while I am writing an open file, sometimes when I try to write a file on the next boot the SD card writing goes extremely slow or will even stop sometimes.

If i plug the SD card into my computer and run CHKDSK i can see 
Convert lost chains to files (Y/N)? 
If i fix that on my computer it is fine again.

My question is there anyway to detect unfinished writes and correct them on the ESP8266. I would like to be able to do it without taking out the SD card

greiman, 2017-11-10:
There is no way to repair a file system on the ESP8266.

closed 2017-11-10 by greiman

schlaegerz, 2017-11-10:
If I really wanted to do this. Do you know of how to to start going about it? Or is there something that is completely blocking it? 

I only want to be able to fix the errors that I caused by not finishing the write, so I should be able to find where the writing to file failed and fix it from there?

greiman, 2017-11-12:
When the power is lost, the record of clusters allocated to the file is lost.  You must write a program like CHKDSK to repair files and the FAT.  It takes a huge amount of RAM to do this.  The ESP8266 is too small.

schlaegerz, 2017-11-14:
For anyone else running into this issue, it helps a lot if you have a long running write that goes over multiple OS loops like i did to make sure you use the flag O_SYNC when opening the file. This greatly reduced the number of corruptions I had.

Also, I was able to get some repairs on the SD card. I loosely based my fix on this library: https://github.com/Gregwar/fatcat/

I iterated over all the files in the stucture and marked which clusters were seen and then iterated over all the clusters in the volumes and removed all the chains that weren't associated with a file.

To do that i had to define two public FatVolume functions.  

```
int32_t FatVolume::FreeChainStartingAtCluster(uint32_t cluster)
{
    return freeChain(cluster);
}
int FatVolume::isFree(uint32_t cluster)
{
    auto lba = m_fatStartBlock + (cluster >> 7);

    cache_t* pc = cacheFetchFat(lba, FatCache::CACHE_FOR_READ);
    if (!pc) {
        return -1;
    }
    if (pc->fat32[cluster & 0X7F] == 0) {
        return 1;
    }

    return false;
}
```

You are limited by how many clusters you can mark as "seen" i used a byte array to minimize the memory, and luckily enough with my system I can be fairly sure I won't have cluster numbers that are  too high.