greiman/SdFat #87: File.printf() question

freeck, 2017-12-03:
I have tried to use the File.printf() method. When I write data to a file using File.write("12..."); the data is written into the data file, but when I use File.printf("Value: %d", 123) ; no bytes are written into the data file.
What do I do wrong?

greiman, 2017-12-03:
The SdFat File class has no printf member.

> Arduino: 1.8.4 (Windows 10), Board: "Arduino/Genuino Uno"
> 
> C:\Users\bill\AppData\Local\Temp\arduino_modified_sketch_85366\ReadWrite.ino: In function 'void setup()':
> 
> ReadWrite:64: error: 'class File' has no member named 'printf'
> 
>  myFile.printf("hi\n");
> 
>         ^

freeck, 2017-12-03:
Thanks for the quick response!
Hum..... I am using Eclipse-Sloeber: class File exposes method printf and compiles without errors. 
I will create a basic app and try again, I let you know.

freeck, 2017-12-04:
Strange....
In case I compile your example for the ESP8266 using ArduinoIDE 1.8.0 it exposes myFile.printf(...); , no error reports!
In case I build the example using Sloeber-IDE; again printf()  is exposed.... AND myFile.printf() works fine!

Concerning my original issue: I am still searching for the root cause, I suppose the usage of much libraries might causes the pain....

greiman, 2017-12-04:
The File class is derived from the Print class.  Many Arduino work-alike systems add the printf() member to the Print class.

I optionally add printf() to file classes in my new [SdFs](https://github.com/greiman/SdFs) library.

freeck, 2017-12-04:
Ok thanks.

closed 2017-12-04 by freeck