greiman/SdFat #88: Esp32 compatibility?

stickbreaker, 2017-12-07:
@greiman I would like to use your excellent library on a ESP32. 

I have a WeMos Battery+BlueTooth (Esp32 Vroom32 V.1 ).  The Sdcard is directly connected to the 3.3V ESP using the ESP32 standard signalling:

[image]

It works with the ESP32 supplied SD library, but when I try one of your examples (SdInfo.ino) it generates this output:
```
SdFat version: 1.0.3

Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.

Assuming the SD chip select pin is: 5
Edit SD_CHIP_SELECT to change the SD chip select pin.

type any character to start
error: cardSize failed
SD errorCode: 0X50,0X0
```
I am using a 1gb MicroSD, Windows 7 shows it as a 1gb FAT Capacity of 922MB, Partition Style MBR

Have your experimented with the ESP32?  Are you planing on supporting it?

Chuck.

greiman, 2017-12-08:
I have no plans to support ESP32.

stickbreaker, 2017-12-08:
@greiman I'm sad to hear that.  Thank you for your work on the Atmel processors. I have enjoyed using your library with Uno's and Mega's.

Chuck.

closed 2017-12-08 by stickbreaker

jasoroony, 2018-02-20:
Have you tried reducing the sd.cardBegin from 50MHz to something smaller (works for me on ESP32 at 10MHz, but I've not tried anything other than SdInfo.ino).

Osvaldolima, 2020-01-28:
como vc fazer  enviar os  dados utilizar o protocolo mqtt

megantaraazharridho, 2021-07-24:
i think you would have adapter module but without shifter.
https://www.aliexpress.com/item/1005002877247771.html