greiman/SdFat #89: Crash when removing file if the SD file is invalid.

schlaegerz, 2017-12-22:
I end up in a Watchdog restart in remove() when there is an invalid file if the next cluster points back to zero. This seems to happen if I lost power while writing occastionally.

I seem to be able to fix it by changing FreeChain:

`bool FatVolume::freeChain(uint32_t cluster) {
  uint32_t next;
  int8_t fg;
  do {
    //yield();
    fg = fatGet(cluster, &next);
    if (fg < 0) {
      DBG_FAIL_MACRO;
      goto fail;
    }
    Serial.println("FREEING" + cluster);
    
    // free cluster
    if (!fatPut(cluster, 0)) {
      DBG_FAIL_MACRO;
      goto fail;
    }
    // Add one to count of free clusters.
    updateFreeClusterCount(1);

    if (cluster < m_allocSearchStart) {
      m_allocSearchStart = cluster;
    }
    cluster = next;
  } while (fg **&& cluster**);

  return true;

fail:
  return false;
}`

greiman, 2017-12-22:
This is not a safe fix.

closed 2017-12-22 by greiman

schlaegerz, 2017-12-22:
I don't get why it is not? All I'm doing is preventing a certain infinite loop. 
Other equal solution I found was to make sure fatPut fails if you try to write an invalid spot.

```
bool FatVolume::fatPut(uint32_t cluster, uint32_t value) {
  uint32_t lba;
  cache_t* pc;

  // error if reserved cluster of beyond FAT
  DBG_HALT_IF(cluster < 2 || cluster > m_lastCluster);
  if(cluster < 2 || cluster > m_lastCluster)
  {
    return false;
  }```

greiman, 2017-12-23:
The problem with simple fixes to a corrupt file system is that clusters may be allocated to multiple files or other problems that can cause many files to be lost or corrupted.

The problem that causes the corrupt file needs to be fixed.

I will make the DBG_HALT error always fatal.