greiman/SdFat #90: SD card not idling after write / close

joko370, 2017-12-22:
Hi,

I am implementing a simple datalogger using the Sparkfun Razor IMU integrated board. 
But the SD card keeps drawing ~14 mA after writing/syncing or closing a file on it.  Data is on the disk and looks fine.  Battery dies out too quick though.   Writing schedule is once ever 30 seconds.  Open file in append mode, storing 9600 bytes, and closing it again.  Current is drawn ever since the first write.  

The Razor board features a SAMD21 unit along with a MPU9250 acc/gyro/mag unit and a micro SD card slot. I used kingston 32 Gb SD card.   
 
To rule out coding issue of my own I tried the datalogger example in the installation.  It shows the same behaviour.   Once it completes <Done> the card keeps drawing current.  cpu, sensors and sd card together 36 mA.   6mA for the sensors,  16 mA for the cpu, 14mA for the SD card.  I understand that after completion of the script, the SD card should go idle which it doesn't.  I am focusing now on that 

I bought 2 new SD cards yesterday.  16 Gb Kingston units.  I formatted them using the SDformatter.  But both have the same issue.   Current is too high.  I am not sure where to look for the problem. 

My main questions are,  
-  Maybe cards are damaged that they dont turn back to idle mode.  How to check that ?
-  Is the library properly initialised for this board ?  How to check that ?

Thanks in advance and best wishes for the holidays to come anyway. 

JosK.

greiman, 2017-12-22:
SD cards go to low power mode after SPI clocks with chip select high.

The SdFat code is at about line 485 of SdSpiCard.cpp

```
void SdSpiCard::spiStop() {
  if (m_spiActive) {
    spiUnselect();
    spiSend(0XFF);  <-- go to low power mode.
    spiDeactivate();
    m_spiActive = false;
  }
}
```
If you use the Arduino SD.h library, this code may be missing.  I did not add it to the old version of SdFat used in SD.h.

The SparkFun examples appear to use SD.h.

joko370, 2017-12-22:
Hi Bill,

Thanks for speedy response.
I was not using the Sparkfun examples but the one distributed with the Sdfat lib.
“dataLogger”  in particular.

But maybe the SD lib is found first and used instead of SDfat.
Will delete it explicitly to see if the examples run different..
It would explain a lot.

Anyway.  Basically executing the  spiStop() should put the card in low power mode.
Will try that in a few minutes to begin with..

Will let you know outcome.

Thanks again.

Jos Koning

ir. Jos Koning | Senior Project Manager | Trials & Monitoring
MARIN | T +31 317 49 32 75 | M +31 652 03 65 29 | J.Koning@marin.nl<mailto:J.Koning@marin.nl> | www.marin.nl<http://www.marin.nl>

[LinkedIn]<https://www.linkedin.com/company/marin> [YouTube] <http://www.youtube.com/marinmultimedia>  [Twitter] <https://twitter.com/MARIN_nieuws>  [Facebook] <https://www.facebook.com/marin.wageningen>
MARIN news: Nederlandse onderzoeksinstellingen bundelen de krachten op het gebied van windmetingen<http://www.marin.nl/web/News/News-items/Nederlandse-onderzoeksinstellingen-bundelen-de-krachten-op-het-gebied-van-windmetingen.htm>

greiman, 2017-12-22:
The datalogger example calls sync() which calls spiStop() so there must be some other problem.

joko370, 2017-12-22:
Hm.  Yes.

Deleted SD lib. ->  no effect.
Deleting SdFat ->  error Sdfat.h not found  as expected.  So that was not it.

Scaled down to bare bones SDcard part to check further.
Empty loop().  Setup()    Calls initSD() straight from your examples.
Then updateSaveFile stripped from my original project.
Directory & file names are now fixed.

The problem occurs usually already when merely opening and closing a file.  (no data write)
With the names fixed.  Running the code twice reopens and closes the same file.
The SD card then does go asleep.
When making a new file and closing it, it does not.    ->  that is strange.  O_CREAT should do that I thought.

When writing something into an existing file.  The card does not turn back to sleep.
That is separate from the Open() Close()  above but both mean that something has to be written onto the card.
So that may be caused by the same issue.

The Close() method however returns true in all cases.
Did I mistake the arguments  of the open() method ?

I added the lines below.
Relevant part is top most.   The more standard initialisation part  is added further down.
My suspicion now is on the open() method.  Does it look sensible ?

Jk.

void updateSaveFile()
{
      PRT.print(savDir); PRT.print("  "); PRT.println(savNme);

      // if not exists already, create new directory
      if (!sd.exists(savDir));
      {   // no error.  ...
         PRT.println("making new directory");
         sd.mkdir(savDir,true);
      };

      PRT.println( sd.chdir(savDir,true) );   // set pwd. Should return 1 for OK !

      // open file with new name
      if (savFil.open(savNme,O_CREAT | O_WRITE  ))     //    Is this what it should be ??
      {
         PRT.println("file create succeeded") ;
         int n = 0;
    //     n = savFil.write("Hello SD2");
         PRT.print(n) ;PRT.println("written");

         PRT.print("file close result:  ");
         PRT.println(  savFil.close()  ) ;  //    -> SD sleeps if it already existed and nothing is written.
      } else
      {
        PRT.println("file create failed") ;
      }
}

void setup()
{
  pinMode(     LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);

  // setup serial outputs
  PRT.begin(19200);
  PRT.println("starting");
  delay(1000);

  initSD();
  delay(1000);
  updateSaveFile();      //  has to be after imu is created...

  PRT.println("Main loop starting");
}

void loop()
{
}

//  ****************************  standard part  ******************************

#include <stdlib.h>
#include <stdio.h>
#include <string.h>
#include <math.h>

#include <SdFat.h>

#define SPI_SPEED SD_SCK_MHZ(50)
#define SD_chipsel          38

// Serial streams
// input buffer for line
char cinBuf[40];
ArduinoOutStream cout(SerialUSB);
ArduinoInStream   cin(SerialUSB, cinBuf, sizeof(cinBuf));

#define          PRT SerialUSB    //   plain Serial USB object for input/output

SdFat                   sd   ;    //  File system object.
SdBaseFile          savFil   ;    //  File to store into

void reformatMsg() {
  cout << F("Try reformatting the card.  For best results use\n");
  cout << F("the SdFormatter program in SdFat/examples or download\n");
  cout << F("and use SDFormatter from www.sdcard.org/downloads.\n");
}

char savDir[80] = "rawData/yyyymmdd             " ;
char savNme[20] = "hhmmss3.dat    "                 ;
int initSD()
{
    // initialise SD card.
    // only when starting up
    // return 0 if OK.
    // -1 when error

    if (!sd.begin(SD_chipsel, SPI_SPEED))
    {
      if (sd.card()->errorCode())
      {
         cout << F(
               "\nSD initialization failed.\n"
               "Do not reformat the card!\n"
               "Is the card correctly inserted?\n"
               "Is chipSelect set to the correct value?\n"
               "Does another SPI device need to be disabled?\n"
               "Is there a wiring/soldering problem?\n");
         cout << F("\nerrorCode: ") << hex << showbase;
         cout << int(sd.card()->errorCode());
         cout << F(", errorData: ") << int(sd.card()->errorData());
         cout << dec << noshowbase << endl;
         return(-1);
      }

      cout << F("\nCard successfully initialized.  But...\n");
      if (sd.vol()->fatType() == 0)
      {
        cout << F("Can't find a valid FAT16/FAT32 partition.\n");
        reformatMsg();
        return(-1);
      }
      if (!sd.vwd()->isOpen())
      {
        cout << F("Can't open root directory.\n");
        reformatMsg();
        return(-1);
      }
      cout << F("Can't determine error type\n");
      return(-1);
    }

    cout << endl;

    return(0);
};

//   *******************************    end of file  ********************************************************

ir. Jos Koning | Senior Project Manager | Trials & Monitoring
MARIN | T +31 317 49 32 75 | M +31 652 03 65 29 | J.Koning@marin.nl<mailto:J.Koning@marin.nl> | www.marin.nl<http://www.marin.nl>

[LinkedIn]<https://www.linkedin.com/company/marin> [YouTube] <http://www.youtube.com/marinmultimedia>  [Twitter] <https://twitter.com/MARIN_nieuws>  [Facebook] <https://www.facebook.com/marin.wageningen>
MARIN news: Nederlandse onderzoeksinstellingen bundelen de krachten op het gebied van windmetingen<http://www.marin.nl/web/News/News-items/Nederlandse-onderzoeksinstellingen-bundelen-de-krachten-op-het-gebied-van-windmetingen.htm>

greiman, 2017-12-22:
I tried the datalogger example on an Arduino Zero.  I set the log interval to 10 seconds (10,000 microseconds).

The Zero has a ATSAMD21G18 CPU.

I connected a bare uSD socket to the SPI header.  I measured the 3.3V current to the SD.

I used a SanDisk 16GB Ultra uSD.

The current before initializing the SD was 0.54 ma.  While logging, the the SD idled at 0.23 ma.

I tried a Samsung 32GB Pro+.  It draws 0.12 ma before initialization but draws 12.17 ma during logging.

The winner was a Transcend 32GB Ultimate 633x.  It idles at 0.09 ma.

I suspect many new cards don't play well in SPI mode.  I don't know of a command in SPI mode to force sleep mode.

greiman, 2017-12-22:
Look at this [article](https://edwardmallon.wordpress.com/2017/05/21/switching-off-sd-cards-for-low-power-data-logging/).

The good old days of low power sleep for SD cards in SPI mode may be gone.

joko370, 2017-12-22:
Tendency seems to be to switch off power via hardware. The integrated razor board does not have that option.
I am not so sharp now.  End of the year party just finished.
Will continue efforts after christmas.  Happy holidays and thanks for assistance.

Verzonden vanaf mijn Samsung Galaxy-smartphone.

MARIN news: Happy holidays & an inspiring 2018<http://www.marin.nl/web/News/News-items/Happy-holidays-an-inspiring-2018.htm>

greiman, 2017-12-23:
I have discovered a possible fix.

SdFat has an option to not check SD card flash programming after a single sector write.  Most modern cards do not require this test since they recover from flash programming errors by replacing the bad flash with spare flash.

This allowed a write speedup since no wait was required.

Some cards seem to not go to sleep unless they receive the command to check programming.

Try changing this symbol to non-zero at about line 542 of SdSpiCard.cpp.

`
#define CHECK_PROGRAMMING 0`
 
I may make the default to non-zero and move the definition to SdFatConfig.h.

greiman, 2017-12-23:
I made check flash programming the default and defined a symbol in SfFatConfig.h.

```
//------------------------------------------------------------------------------
/**
 * If CHECK_FLASH_PROGRAMMING is zero, overlap of single sector flash 
 * programming and other operations will be allowed for faster write
 * performance.
 *
 * Some cards will not sleep in low power mode unless CHECK_FLASH_PROGRAMMING
 * is non-zero.
 */
#define CHECK_FLASH_PROGRAMMING 1
```

joko370, 2017-12-23:
Good evening
Cant check right now bcs family obligations. Will try out earliest but may be
Second day xmas.
Impressive work btw, both the sdfat library itsekf and your support.

I am fairly new to SD storage via microcontroller.  Feels a little daunting to diginto. I did do some nosing in the sd physical data layer description this afternoon. In spi mode the sd cards should go to idle via CMD0.  I can imagine that could conflict with the timing and bufferinf in sdfat.  Is there any merit in pursueing such a route if CHECK_PROGRAMMING does not do the trick ?

Verzonden vanaf mijn Samsung Galaxy-smartphone.

greiman, 2017-12-23:
CMD0 resets the SD.  All files must be closed since all state is lost.

The idle mode for CMD0 is not the same as the low power sleep mode.  CMD0 boots the internal controller and waits for the next command in the initialization sequence.

You would need to reinitialize the SD with another begin() call, reopen and reposition your file.

joko370, 2017-12-26:
Good day Bill,

CHECKED_FLASH_PROGRAMMING did indeed solve the issue.
Thanks for the support and best,
Jk

closed 2017-12-26 by joko370