greiman/SdFat #91: Weird behaviour while writing certain bytes to binary files

alieslam, 2018-01-15:
## Basic Infos
**Hardware**

Hardware: NodeMCU 1.0 ESP-12E
Core Version: 2.3.0

## Description
Writing (copying) a certain array of 30 bytes (testarr1)  to a binary file (test1) fails and the bytes in (test1) binary file is different after the first 8 bytes then the rest are completely messed up , although i have added another array of 30 bytes (testarr2) in the same sketch and copied them to another binary file (test2) and it seems to be working like charm. It seems that it doesn't work with testarr1 sequence (still dunno why) but i have tried lots of different bytes sequnce and they all work perfectly. 

The conclusion is that there is something to do with the sequence of bytes in testarr1 which fails copying them to test1 binary file, but i didn't reach to any insights (till now).

## Settings in IDE
Module: NodeMCU 1.0 ESP-12E
Flash Size: 4 MB
CPU Frequency:80Mhz
Upload Using:	Serial
Reset Method:	?ck / nodemcu?

## Sketch
#include <SPI.h>
#include "SdFat.h"
SdFat SD;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  do {
    Serial.println("Mount SD card");
  } while (!SD.begin(4, SD_SCK_MHZ(4)));
  File test1 = SD.open("test1", FILE_WRITE);
  File test2 = SD.open("test2", FILE_WRITE);
  uint8_t testarr1[] = {0x01, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x65, 0xf7, 0xe0, 0x8e, 0xa8, 0xef, 0x74, 0xb0, 0x34, 0x85, 0x36, 0x9f, 0x35, 0x80, 0xf8, 0x5f, 0x6a, 0xd5, 0x3d, 0x76, 0x31, 0x37, 0x19};
  uint8_t testarr2[] = {0x02, 0x00, 0x00, 0x00, 0x01, 0x93, 0x76, 0x05, 0xf7, 0xf8, 0x08, 0xa7, 0x6c, 0x93, 0xe0, 0xc3, 0xd5, 0x9e, 0x31, 0x14, 0x83, 0x9a, 0xf4, 0xc3, 0x50, 0xc0, 0x66, 0x41, 0xd6, 0x6b};
  test1.write(testarr1, 30);
  test1.close();
  test2.write(testarr2, 30);
  test2.close();
  Serial.println("End");
}

void loop() {
  // put your main code here, to run repeatedly:

}

## Debug Messages
None

alieslam, 2018-01-15:
Update: tested the same sketch (changed SD chip select to pin 8 in sketch) with Arduino UNO board and still the same result.

greiman, 2018-01-15:
I ran the following version on a Adafruit HUZZAH ESP8226.
```
#include <SPI.h>
#include "SdFat.h"
#define CS_PIN SS
SdFat SD;
void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
  // Wait for USB Serial 
  while (!Serial) {
    SysCall::yield();
  }
  Serial.println("Type any character to start");
  while (!Serial.available()) {
    SysCall::yield();
  }
//  do {
//    Serial.println("Mount SD card");
//  } while (!SD.begin(4, SD_SCK_MHZ(4)));

  if (!SD.begin(CS_PIN, SD_SCK_MHZ(4))) {
    Serial.println("begin failed");
    return;
  }
  File test1 = SD.open("test1", FILE_WRITE);
  if (!test1) {
    Serial.println("open test1 failed");
    return;
  }
  File test2 = SD.open("test2", FILE_WRITE);
  if (!test2) {
    Serial.println("open test2 failed");
    return;
  }
  uint8_t testarr1[] = {0x01, 0x00, 0x00, 0x00, 0x00, 0x01, 0x01, 0x65, 0xf7, 0xe0, 0x8e, 0xa8, 0xef, 0x74, 0xb0, 0x34, 0x85, 0x36, 0x9f, 0x35, 0x80, 0xf8, 0x5f, 0x6a, 0xd5, 0x3d, 0x76, 0x31, 0x37, 0x19};
  uint8_t testarr2[] = {0x02, 0x00, 0x00, 0x00, 0x01, 0x93, 0x76, 0x05, 0xf7, 0xf8, 0x08, 0xa7, 0x6c, 0x93, 0xe0, 0xc3, 0xd5, 0x9e, 0x31, 0x14, 0x83, 0x9a, 0xf4, 0xc3, 0x50, 0xc0, 0x66, 0x41, 0xd6, 0x6b};
  if (test1.write(testarr1, 30) != 30) {
    Serial.println("write test1 failed");
    return;
  }
  if (!test1.close()) {
    Serial.println("close test1 failed");
    return;
  }
  if (test2.write(testarr2, 30) != 30) {
    Serial.println("write test2 failed");
    return;
  }
  if (!test2.close()) {
    Serial.println("close  test2 failed");
    return;
  }
  Serial.println("End");
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
The sketch printed:
> end

The files contained:
```
F:\>xxd test1
00000000: 0100 0000 0001 0165 f7e0 8ea8 ef74 b034  .......e.....t.4
00000010: 8536 9f35 80f8 5f6a d53d 7631 3719       .6.5.._j.=v17.

F:\>xxd test2
00000000: 0200 0000 0193 7605 f7f8 08a7 6c93 e0c3  ......v.....l...
00000010: d59e 3114 839a f4c3 50c0 6641 d66b       ..1.....P.fA.k
```

greiman, 2018-01-15:
I used  Arduino IDE 1.8.5 with ESP8266 board package 2.4.0.

alieslam, 2018-01-15:
@greiman 
Thanks for your quick response..! 
I have updated my core to 2.4.0 and tested with both your sketch and my old one, and everything is working perfectly which made me skeptical about my hex editor and (bam..!) It was Notepad++ hexEditor plugin which displays faulty readings (**GUYZ AVOID USING NOTEPAD++ HEX EDITOR**), i have been using this plugin for a long time and nothing liked that popped up before, but i shouldn't have made assumptions (lesson learned).
Thanks again @greiman  for your time, closing this Issue.

closed 2018-01-15 by alieslam