greiman/SdFat #92: After many hours I'm completely stumped as to what might be going on.

Brett-Howard, 2018-01-22:
First off thank you for the huge effort that you've put in to produce this fabulous library!  

I'm working on a wind instrument and GPS logger.  The project at the moment deals with a few files.  It first powers up and deletes a stats file (from the last execution).  It then opens up the configuration file (if present) and reads that in.  If its not present it makes a default one.  

Then in the loop about once per minute it writes to a newly created stats file.  

So far this all works.  

Then I took the code from my fully working other GPS logger project and tried to integrate it into this one.  

The GPS logger opens the file steps backward 28 bytes with a seekSet() then writes a new record and closes the file.  This function gets called once per second.  I know that closing the file that often is a bit inefficient but it works quite well in my dedicated GPS logger project and allows one to just pop the card out whenever and the files are always in tact.  

I've been finding in this project that this additional once per second log is regularly crashing in the function that does the writing.

`static void WriteGPXLog()
{ 
  // Log the fix information if we have a location and time
  if (globalFix.valid.location && globalFix.valid.time) {
    char date1[22];
    snprintf(date1, 22, "%4d-%02d-%02dT%02d:%02d:%02dZ", globalFix.dateTime.full_year(), globalFix.dateTime.month, 
            globalFix.dateTime.date, globalFix.dateTime.hours, globalFix.dateTime.minutes, globalFix.dateTime.seconds);
    
    if(!gpsLog.isOpen())
      gpsLog.open(filename, O_WRITE);
      
    gpsLog.seekSet(gpsLog.fileSize() - 28)
    gpsLog.print(F("\t\t<trkpt lat=\""));

    if (globalFix.valid.location) {
      gpsLog.print(globalFix.latitude(), 7);
      gpsLog.print(F("\" lon=\""));
      gpsLog.print(globalFix.longitude(), 7);
      gpsLog.print(F("\">\n"));
    }

    if (globalFix.valid.altitude) {
      gpsLog.print(F("\t\t\t<ele>"));
      gpsLog.print(globalFix.altitude(), 2);
      gpsLog.println(F("</ele>"));
    }

    if (globalFix.valid.time) {
      gpsLog.print(F("\t\t\t<time>"));
      gpsLog.print(date1);
      gpsLog.print(F("</time>\n"));
    }

    if (globalFix.valid.speed) {
      gpsLog.print(F("\t\t\t<speed>"));
      gpsLog.print(globalFix.speed());
      gpsLog.print(F("</speed>\n"));
    }

    if (globalFix.valid.hdop) {
      gpsLog.print(F("\t\t\t<hdop>"));
      gpsLog.print(float(globalFix.hdop)/float(1000), 3);
      gpsLog.print(F("</hdop>\n"));
    }

    if (globalFix.valid.heading) {
      gpsLog.print(F("\t\t\t<course>"));
      gpsLog.print(globalFix.heading());
      gpsLog.println(F("</course>"));
    }

    gpsLog.print(F("\t\t\t<sat>"));
    gpsLog.print(globalFix.satellites);
    gpsLog.println(F("</sat>"));

    gpsLog.print(F("\t\t</trkpt>\n"));

    //replace closing tags
    gpsLog.print(F("\t</trkseg>\r\n</trk>\r\n</gpx>\r\n"));

    gpsLog.close();
    blip(RED_LED_PIN, 1, 20);
  }
}`

That above is the function that is doing the writing.  I've tried putting while()'s around the open and the close so that it will keep trying if things fail and that did make it last a little longer on average.  

I've also set the file to only be called every 5 seconds and I end up with no improvement in robustness.  

I've also recently made sure that this has its own SdFile object as previously I was just reusing the object that I used to handle the configuration file.

If I comment out the file operations in this function the system holds together and works great for multiple days on end.  If I comment out everything except for the file open and the file close it still crashes about the same.  I even added 150mS of delay between them just incase it was that open/close were happening to close to one another...

The only difference between where this code is used and working well is that it is working well on a Adafruit Adalogger 32u4 but its not working in this larger project using an adalogger m0.  Both systems call this function once per second.  

I've also tried slowing down the sd interface.  Currently I'm still having issues with my initialization code as:
`if (!sd.begin(SD_CHIP_SEL, SD_SCK_MHZ(1))) {`
The 32u4 version just has the chip select defied and no clock speeds are defined.

I should add that there is also an RF95 LoRa radio attached to the SPI port on this project.  But that has no issues with any of my other file I/O in the design...  Really at my wits end and have around 20 hours into trying things and just can't come up with anything.  

Last clues that I have are that I've got a gesture sensor that I've got hooked up to an ISR.  When the system crashes I find that I can get the ISR to fire one time.  As I've seen recommended I put:
`Serial.println(sd.card()->errorCode(), HEX);`
into the isr handler.  Then when things crash I wave my hand over it and I can get the error if there is one...  So far I've seen 0x57 once and I get a few 0x30 and 0x32...  Seems like errors in the 30's are the most common however 0 is also often there when things hang up.  I've also tried multiple Sd cards all SanDisk branded.  

Another difference is that this is the first time in this project that the system has had two files open at once but I've seen where you've replied that the only limitation on the number of files one can have open is limited by system memory.  I've also checked that and the system has a steady 25,383 bytes free while this loops runs.  Not sure what else I'm leaving on the table or forgetting to mention that I've tried and/or am doing...

Brett-Howard, 2018-01-22:
With 300mS put between the completion of writing the GPX log entry and the start of processing radio traffic things seem to hold together better than they have been (haven't had long to test).  But 50mS doesn't seem to be enough.  Unfortunately adding that much delay into my loop really makes the project sluggish so I'll have to make a non blocking delay that only happens after the GPX log is written.  Here's hoping that this is a sustainable workaround.  Strangely though there is a write to another file just above the GPX file write.  This doesn't write near as much data and it only does so once a minute though...

Brett-Howard, 2018-01-23:
Ok I'm now seeing this in the RadioHead library:
The RH_RF95 driver uses interrupts to react to events in the RFM module,
/// such as the reception of a new packet, or the completion of transmission
/// of a packet.  The RH_RF95 driver interrupt service routine reads status from
/// and writes data to the the RFM module via the SPI interface. It is very
/// important therefore, that if you are using the RH_RF95 driver with another
/// SPI based deviced, that you disable interrupts while you transfer data to
/// and from that other device.  Use cli() to disable interrupts and sei() to
/// reenable them.

I've now disabled interrupts in all functions that do file I/O but I still have problems.  Makes me think that the SdFat library is doing some caching that is making things happen later and the radio and the SD card are stomping on one another.....

Brett-Howard, 2018-01-23:
Editing the SdSpiDriver.h file to include noInterrupts() in the activate() and interrupts() in the deactivate() functions seems to have the system working reliably for an hour so far.  It might be really nice to have an option in the SdFatConfig.h file to disable interrupts during file I/O for situations such as this.

greiman, 2018-01-23:
You must use [SPI transactions](https://www.arduino.cc/en/Tutorial/SPITransaction) to protect interrupts.

> With the new SPI library, configure each SPI device once as an SPISettings object. Also, if that device will be called from an interrupt, say so with SPI.usingInterrupt(interruptNumber). To communicate with a specific SPI device, use SPI.beginTransaction which automatically uses the settings you declared for that device. In addition, it will disable any interrupts that use SPI for the duration of the transaction. Once you are finished, use SPI.endTransaction() which re-enables any SPI-using interrupts. 

Fix the radio driver!

closed 2018-01-23 by greiman

Brett-Howard, 2018-01-24:
Thank you for your time! I'll start investigating what can be done to help improve the other library...

JBL-mach1, 2021-02-01:
Hi, I am running into the same issue. I am trying to return to using the radio after logging to the SD card. I am using the RH_RF95 driver, the RHReliableDatagram manager, and SdFat. I was wondering if you had any success using SPI transactions? Thank you.

Brett-Howard, 2021-02-01:
I tried different versions of SdFat and several different versions of the
radio code...  in the end I ended up needing to just kludge a lock and
unlock into the version of SdFat that I used...  It was just a simple
personal project and it's been working for years since.

~Brett

Brett-Howard, 2021-02-01:
Finally found the comment I left to my self about this:

This seems to hold together quite well (been working for an hour now). I
needed to add noInterrupts(); to the activate() function and interrupts();
to the deactivate() function in SdSpiDriver.h of the SdFat library in order
to get this all going.

greiman, 2021-02-01:
Do not use noInterrupts()/ interrupts() in SdFat or other libraries to fix problems in  libraries that use SPI in interrupt routines.  You will make various parts of the Arduino system unstable if you use libraries that don't obey the SPI protocol.

> usingInterrupt()
> Description
> 
> If your program will perform SPI transactions within an interrupt, call this function to register the interrupt number or name with the SPI library. This allows SPI.beginTransaction() to prevent usage conflicts. Note that the interrupt specified in the call to usingInterrupt() will be disabled on a call to beginTransaction() and re-enabled in endTransaction(). 
> 

SdFat does not use interrupts and should not have noInterrupts()/ interrupts() added.
 See this:

https://www.arduino.cc/en/Reference/SPIusingInterrupt

Brett-Howard, 2021-02-01:
Yes the problem probably more lies in the radio code and I tried to fix it
"right" for quite some time and the only thing that got me over the hurdle
and to a working solution was what I did. I'm sure there is more that could
be done. But this worked well in my situation.

~Brett

greiman, 2021-02-01:
Too bad, Paul Stoffregen of Teensy fame worked on the SPI transactions to provide a safe optimal way to use SPI in interrupt routines.  I expected many libraries would never be updated but updated my libraries to conform to SPI transactions.

Brett-Howard, 2021-02-01:
I do not disagree and I read through and started to understand some of this
but I was unable to find the solution in the radio head code and making a
slight tweak to the SdFat code was what got things working.  I realize this
is more of a personal failing and a lack of time but its only a report of
what allowed "me" to get through to the other side.  Perhaps this
information will help this person find the real issues in the radio head
project but for me the SdFat project was much easier to follow.