greiman/SdFat #97: Question related to :  error("Please delete existing " TMP_FILE_NAME);

BeinerF, 2018-02-17:
Hi,
first of all thanks for this great library!!
Many of my projects are based on the great LowLattencyLogger example and I think it is time for me to dive a bit deeper in the code to better understand all the ins and outs.

To the point, I can't get my head around one important detail in the recoverTmpFile() function.

- What can lead to the need for a user to manually delete an existing tmp file ? -> error("Please delete existing " TMP_FILE_NAME);
- Is there a smart way to deal with this exception without the need for manual deletion?

It could be useful to add an additional explanatory comment around this code section. (Please ignore if  I am just the only one who does not get it.)

Any hint would be much appreciated. cheers

greiman, 2018-02-17:
I ask the user if I should recover  a tmp file.  If the user does not type 'Y', I can't go on without writing over the file so I ask the user deal  with the file and quit.

If I find a problem with the tmp file in recoverTmpFile(), I quit and ask the user to remove the tmp file.

I am reluctant to just delete the file.

BeinerF, 2018-02-17:
Thx, this answered my question, my thinking was that there is a deeper technical  underlying reason that calls for a manual user intervention. 
I will modify it (just automatically delete the file) for my specific application -- a standalone logging situation that should be stable and recover automatically with as little user intervention as possible.
cheers

closed 2018-03-01 by greiman