greiman/SdFat #98: SD errorCode: 0X5A,0X2 when running TeensySdioDemo with SdFatSdioEX

dkadish, 2018-02-20:
* Arduino: 1.8.1
* SdFat: 1.0.5
* Board: Teensy 3.6 + Teensyduino

I am getting the following error when I try to run the demo. It occurs whenever the demo is run with option `1`, but here is the output when `2` is run first, just for reference to show that `2` works.

```
SdFatSdioEX uses extended multi-block transfers without DMA.
SdFatSdio uses a traditional DMA SDIO implementation.
Note the difference is speed and busy yield time.

Type '1' for SdFatSdioEX or '2' for SdFatSdio

size,write,read
bytes,KB/sec,KB/sec
512,230.41,1309.11
1024,386.75,1965.46
2048,781.68,3241.64
4096,1293.48,5107.68
8192,1908.44,6852.60
16384,2699.42,8284.37
32768,3194.98,9313.04

totalMicros  103486477
yieldMicros  102900496
yieldCalls   81179
yieldMaxUsec 462711
kHzSdClk     22500
Done
Type '1' for SdFatSdioEX or '2' for SdFatSdio

size,write,read
bytes,KB/sec,KB/sec
512,error: write failed
SD errorCode: 0X5A,0X2
```

This appears to be occuring at line 94 and as far as I can tell, the first part of the error relates to a `SD_CARD_ERROR_WRITE_TIMEOUT`. Is that right? Do I need to lengthen the timeout (and if so, where do I do that? I didn't see a define for it), or is there some other issue? https://github.com/greiman/SdFat/blob/e4190e50a4ed83c73fb046b7fce2694e8cfe2ddb/examples/TeensySdioDemo/TeensySdioDemo.ino#L94

Thanks!

dkadish, 2018-02-20:
**Update**

I tried a couple of things to fix this.

_Adjusting Timeouts_
I found the timeouts and adjusted them, but this didn't seem to help (though I'm not 100% positive that the library recompiled properly)
https://github.com/greiman/SdFat/blob/11d6d9ccf4aafce4ec22c21c81d6e3bcd58da74b/src/SdCard/SdInfo.h#L121-L129

_Different SD Card_
I swapped the original SD card for a new one and, voila, *_it worked_*. Seems like that was the issue all along.

*Original (not working)*: SanDisk 2GB MicroSD. No other visible markings.
*New (working)*: Kingston M4GB MicroSDHC Speed Class 4

closed 2018-02-20 by dkadish

greiman, 2018-02-20:
I just checked with a Samsung card and it worked so I was about to suggest trying another card.  I think some old cards fail at higher speeds.  Originally, cards only supported 25 MHz clock.

All microSD cards should run at 50 MHz but some older cards fail.