greiman/SdFat #101: Calls to card() cause a call to syncblocks, which stops multiblock commands

victorpv, 2018-03-31:
I have this problem with the SDIO in the F1. I suspect is the same issue in the teensy, although I don't have a teensy board to test.
The problem is that calls to card() to get the card driver pointer produce a call to syncblocks.
Due to that, multiblock commands are stopped and the driver returned to idle state, so the next sequential read or write will need a new read/write command. It kills the advantage of the Ex class.
In this example card() is called each time yield is called, and as a result the multiblock transfer terminate:
https://github.com/greiman/SdFat/blob/e4190e50a4ed83c73fb046b7fce2694e8cfe2ddb/examples/TeensySdioDemo/TeensySdioDemo.ino#L39

Why is it necessary to call syncblocks just to get the pointer?
In my case to work around that, I use a pointer in the sdio test sketch. The pointer is set once during setup, and the yield function use the pointer instead of calling card() each time, but I wonder why the call to syncblocks.

```
setup(){
...
cardPtr = sd.card();
...
}

yield(){
... cardPtr->isBusy();
}
```

greiman, 2018-04-01:
It may be possible to remove the syncBlocks() call. The first versions of the EX idea had problems with how chip select was handled so I added the syncBlocks() call.  Also calls to functions like  readCID() don't work correctly unless you call syncBlocks().

I would like to remove the card() member function and replace it with calls like sd.cardSize(), sd.cardType().
 sd.isBusy(), sd.readCID(), sd.readCSD(), sd.readOCR().

I will first do this in [SdFs ](https://github.com/greiman/SdFs)to see what happens.

victorpv, 2018-04-02:
Perhaps would be better to have the call to syncBlocks in the functions that need it, such as readCID, but not in card, so one could call isbusy() without stopping the multiblock?
Did you test yielf with the Ex class in a teensy SDIO? from reading the code, or would think the Ex class would not performn any better than the normal class, but perhaps I am missing something and there is another way other than my workaround of using a pointer.

greiman, 2018-04-02:
> Did you test yielf with the Ex class in a teensy SDIO? from reading the code, or would think the Ex class would not performn any better than the normal class, but perhaps I am missing something and there is another way other than my workaround of using a pointer.

Looks like I override card() in SdFatSdioEX at Line 391 of SdFat.h
```
  /** \return Pointer to SD card object */
  SdioCardEX* card() {
    return &m_card;
  }
```
I don't remember the history but the Teensy SdFatSdioEX is fast because there is no syncBlocks call.

Here is the parent class at line 90 of SdFat.h

```
  /** \return Pointer to SD card object */
  SdDriverClass *card() {
    m_card.syncBlocks();
    return &m_card;
  }
```

victorpv, 2018-04-02:
Thanks! I missed this override
`  /** \return Pointer to SD card object */
  SdioCardEX* card() {
    return &m_card;
  }`

That's the key, in the F1 SDIO driver I was using multiblock r/w, but not using the SdioCardEX class, so it was using the normal card() with syncblocks.
I'll rewrite parts of the driver so it works in non-multiblock mode when not using the EX class, and only use multiblock when using EX. That should solve the problem with that.

greiman, 2018-04-02:
On Teensy some users like non-EX mode with huge DMA transfers.  They get the best overlap of compute and DMA.  Teensy 3.6 had 256 KiB of SRAM.

I just tested the Teensy SDIO example.  Size is the size of the buffer in a read/write call.

In EX mode it does a bit under 20 MB/sec write and a bit more for read.  It uses lots of CPU since I can't use DMA due to an errata in the SDIO controller.

Much of the non-EX DMA test is slow but most of the time could be recovered with yield.  The test took about 42 seconds but 41 seconds could be recovered.

```
SdFatSdioEX uses extended multi-block transfers without DMA.
SdFatSdio uses a traditional DMA SDIO implementation.
Note the difference is speed and busy yield time.

Type '1' for SdFatSdioEX or '2' for SdFatSdio

size,write,read
bytes,KB/sec,KB/sec
512,19519.47,20269.98
1024,19264.36,20393.17
2048,19736.88,20528.71
4096,19748.50,20607.49
8192,19842.90,20594.59
16384,19742.13,20777.72
32768,19958.62,20771.64

totalMicros  5838740
yieldMicros  263041
yieldCalls   183
yieldMaxUsec 4126
kHzSdClk     48000
Done
Type '1' for SdFatSdioEX or '2' for SdFatSdio

size,write,read
bytes,KB/sec,KB/sec
512,619.12,2174.87
1024,1165.77,3178.05
2048,2206.78,5447.83
4096,3986.69,8287.49
8192,4190.82,9238.76
16384,7512.02,13674.01
32768,12743.26,15782.69

totalMicros  41528702
yieldMicros  41125939
yieldCalls   78576
yieldMaxUsec 24655
kHzSdClk     48000

```

victorpv, 2018-04-02:
In the F1 I can use both DMA with multiblocks, and recover the yield time. The only problem I have had is that the board I am using for testing doesn't not have pull up resistors in the data lines and doesn't seem to work fine with some cards over 12Mhz. The ram is at max 64KB in the F1, but even with small blocks I get almost full speed (12Mhz * 4bit wide ~ 6MB):
`Card type: SDHC

Manufacturer ID: 28
OEM ID: BE
Product: SDU16
Version: 1.0
Serial number: 2f000000
Manufacturing date: 4/2012

cardSize: 16012.80 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true
OCR: c0ff8000

Volume is FAT32
blocksPerCluster: 64
clusterCount: 488416
freeClusters: 488413
freeSpace: 16004.32 MB (MB = 1,000,000 bytes)
fatStartBlock: 8752
fatCount: 2
blocksPerFat: 3816
rootDirStart: 2
dataStartBlock: 16384

FreeStack: 55503
######################################################

Type any character to start

>>Send to COM3: "1"<<
Test started - please wait, it may take up to 3 minutes

size, write, read
bytes, KB/sec, KB/sec
512, 4507.83, 5131.75
1024, 5066.29, 5511.97
2048, 5351.16, 5658.31
4096, 5484.59, 5736.69

totalMicros  1589389
yieldMicros  1434590
yieldCalls   7696
yieldMaxUsec 2944
kHzSdClk     12000
Done`

victorpv, 2018-04-13:
Bill, I have modified my libmaple F1 driver so it works with SdSdioEX. Now I have a couple of questions.
First, I used multiblock r/w by default, even when not using SdSdioEX. The only problem was with the one reported in the first post.
Should I remove that from the normal readblock and writeblock functions, so they send a stop command after completing the number of blocks there were called to transfer? 
In other words, is there a reason to not use the multiblock modes with sequential blocks all the time even when not using SdSdioEX class?
In the F1 I am able to use DMA and multiblocks together, so it's not like in the teensy that you use either one or the other.
I guess with multiblock transfers there is a bit bigger chance of corruption in case of powerloss, since the card is not sent the stop command right away to the write. Is that a concern?

And my second question is, is there a reason you didn't implement some sort of retry mechanism? in the F1 board I am using, when using 4bit mode at high speeds, some times I get CRC errors. In part seems to be a board design issue, but the board was designed for F4, which can have internal pullups in the SDIO lines, while the F1 doesn't have pull ups, and the board doesn't have pull ups either. Result I get some times CRC errors, and other times "start bit not detected in all data lines". So for my own benefit, I added code to retry up to 3 times if a read or a write fails, and seems to work fine. I was wondering if it's ok to leave it there, of you are opposed to retries for some reason, or you have or plan to add some retry mechanism in the sdFat library.

If you can give me some indication what's best to do in those two subjects, it would be great help.
Thanks.

greiman, 2018-04-13:
> In other words, is there a reason to not use the multiblock modes with sequential blocks all the time even when not using SdSdioEX class?

I think it will be rare for users to have problems if you always use multiblock mode.  

I am always amazed at what some users do.

There may be problems for users who do raw I/O to the SD.  The Idea is that you use createContiguous() to make a large contiguous file and contiguousRange() to find the location on the SD.

You then use raw writes to blocks.  This way you can be sure that write of a single block at any location will fast.  If you use multiblock mode, this won't be true since read stop causes internal SD buffers to be written.

My drivers have a problem with other commands like readCID() but these can be fixed.

> And my second question is, is there a reason you didn't implement some sort of retry mechanism? 

I don't like retries. The CRC on SD commands is only 7-bits.  This means there is a fair probability, 1/128, that a bad address in a write command will not be caught.

Maybe the best answer is to offer a config option for retries.